Kinovarobotics/Kinova-kortex2_Gen3_G3L #99: Modbus with Gen 3 Lite

TomSAVU, 2022-01-29:
If I want to communicate with a Gen 3 Lite via Modbus, Joint 7 in the ICD means the gripper?

felixmaisonneuve, 2022-01-31:
Hi @TomSAVU,

Joint 7 is used for joint 7 on Gen3 arms.

With modbus, you can move the arm in Cartesian mode, which means you control the gripper position and its orientation, but you **cannot** open/close the gripper directly. 

Does that answer your question?

Best,
Felix

TomSAVU, 2022-01-31:
Thanks @felixmaisonneuve ,
Yes, unfortunately it answers my question.
I wonder why to control it using Modbus if I can't control the gripper.
Hopefully this would be added soon ...

felixmaisonneuve, 2022-01-31:
Actually, I was talking with someone and this feature might be already added, but not documented on the github page. 
Because of that, I thought we couldn't open/close the gripper. 

We will test it on a Gen3 Lite arm and I will get back to you in the next couple of days about it

TomSAVU, 2022-02-01:
Thanks for the good news @felixmaisonneuve !
Looking forward to your answer!

felixmaisonneuve, 2022-02-02:
Hi @TomSAVU,

We have tried it and it is possible to move the gripper using modbus. 

You can create an action (or multiple ones) in the webapp and use the action_id in modbus. 

In modbus, you would need to write 0 to address 202 to set the control mode to `Action`, then write the action_id of the gripper action to address 201 to set program to be executed. Finally, you write 1 to address 200 to start program.

In other words, you cannot control the gripper directly from modbus, but you can create an Action that opens/closes your gripper and use modbus to execute that action. 

Regards,
Felix

TomSAVU, 2022-02-04:
Thanks a lot @felixmaisonneuve ,
This really helps!

closed 2022-02-04 by TomSAVU