Kinovarobotics/Kinova-kortex2_Gen3_G3L #100: DATABASE_ERROR - description: no type information

phuicy, 2022-03-04:
## Description

I have upgraded the API of the robot to v2.3 from v2.2, and now get this fault on connection.

```
terminate called after throwing an instance of 'Kinova::Api::KDetailedException'
  what():  Device error, Error sub type=DATABASE_ERROR => <srv: 2, fct: 44, msgType: 3>
description: no type information
```

## Version

[image]

phuicy, 2022-03-04:
```
  auto action_type = Base::RequestedActionType();
  action_type.set_action_type(Base::REACH_JOINT_ANGLES);
  auto action_list = base->ReadAllActions(action_type);
```

failing to get Action list

felixmaisonneuve, 2022-03-04:
Hi @phuicy,

A factory reset of your arm should fix your issue. 
You can do this from the web app in Configurations->Robot->Arm->Base->Reset Factory Settings

Just be carefull, it will wipe all the protection zones and actions that are present. 

Regards,
Felix

phuicy, 2022-03-04:
Initial testing shows this as resolved.

thanks,
Guy

closed 2022-03-04 by phuicy