Kinovarobotics/Kinova-kortex2_Gen3_G3L #101: Euler Angle convention confusion.

Tirine, 2022-03-20:
Hello!

I read your docs a while ago and as I hadn't used so much Euler angles before, but I saw:
 > Note: Orientations are defined using an z-y-x Tait-Bryan extrinsic convention.

Good! Looking at Wiki and some articles with Eigen cpp library how to apply and get Euler angles to/from rotation matrix and quarternions, that means, in the fixed frame of the robot, Rotate on Z, then on Y, then on X, right? Or, as it's the same as X-Y-Z intrinsic, in the frame of the moving object, Rotate on X, then Y', then X'', riiight?
But testing this didn't work.. Today I went back to the docs again, thinking I must have read wrong.
No I didn't read wrong, but I didn't read the whole paragraph.... 
 > That is, rotation about the fixed x-axis, followed by rotation about the fixed y-axis, followed by rotation about the fixed z-axis.

 Isn't that, fixed axis, aka the robot base reference frame, meaning it's "X-Y-Z Extrinsic"? (Aka Intrinsic Z-Y-X ? )
 I can't say I'm knowledgeable of Euler angle conventions, but unless I completely misunderstood Euler angle conventions, this is confusing.. If I'm wrong, could you try to clarify this line in the docs a bit more?

Tirine, 2022-03-21:
> After that, I think you got confused and swapped everything around. You are talking about X-Y-Z Extrinsic (Aka Intrinsic Z-Y-X), but the documentation clearly states Z-Y-X Extrinsic.

Yes, that's where I'm confused, but because that's what your documentation says:
> That is, rotation about the fixed x-axis, followed by rotation about the fixed y-axis, followed by rotation about the fixed z-axis.
That sounds like X-Y-Z Extrinsic to me. Even though the very sentence before says Z-Y-X Extrinsic.

felixmaisonneuve, 2022-03-21:
Ok, I understand your confusion. This sentence is there to describe to manipulation needed to your actual orientation to apply your rotation.

To compute a XYZ Extrinsic, you need to first multiply your x-axis rotation matrix, followed by your y-axis rotation matrix and finally your z-axis rotation matrix (`Rz * Ry * Rx`).

Tirine, 2022-03-21:
> To compute a XYZ Extrinsic

Which means you're using XYZ Extrinsic, not using ZYX Extrinsic? So the first part of the phrase in the docs is wrong?

Also I did do some testing last week, I can confirm in later this week again:
I set the robot to a position with rotation [90, 0, 45], which if it's in Z-Y-X Extrinsic, the pose's Z Axis should still be along the robots negative Y Axis. But if I apply that pose on the real robot, the Z axis is along the diagonal 45 degree of the robots X and -Y plane, which I would expect from a Z-Y-X Intrinsic rotation, or as your write in the second phrase of your description, X-Y-Z Extrinsic rotation...

And last maybe it's converted somewhere else, but in the ROS Kortex, you guys use [KDL::Rotation::GetEulerZYX](https://docs.ros.org/en/indigo/api/orocos_kdl/html/classKDL_1_1Rotation.html#ae927b1f8077d6e827ea2968ca20af136), which is a function using ZYX Intrinsic, not Extrinsic:
>GetEulerZYX gets the euler ZYX parameters of a rotation : First rotate around Z with alfa, then around the new Y with beta, then around new X with gamma.

 but you use your values from the robot straight to it, which if it works fine, means the values you're using are into the function is ZYX Intrinsic?

felixmaisonneuve, 2022-03-22:
Hi @Tirine,

We are using ZYX Extrinsic. I wrote the wrong thing in my previous comment, I meant "XYZ Intrinsic" and not "XYZ Extrinsic" (I corrected it). 

For your testing, what you got is Z-Y-X Extrinsic. 
A Z-Y-X Extrinsic rotation is the equivalent of starting at [0,0,0], then you apply the z-axis rotation [0,0,45], then the y-axis rotation [0,0,0] (does nothing, so I will skip this), then X-axis rotation [90,0,0] while always keeping your fixed XYZ system. 
(Blue is the Z-axis, Green is Y-axis and Red is X-axis)
We start at [0,0,0]
[image]

we do a 45 degrees rotation on the Z-axis
[image]

then we do a 90 degrees rotation on the X-axis
[image]

Which brings the Z-axis alongs the diagonal 45 degree of the robots X and -Y plane.

If it was XYZ Extrinsic, we would have to rotate on X first
[image]

Then rotate the Z-axis by 45 degrees, which would give something like that
[image]

ROS Kortex does use `GetEulerZYX` (which is ZYX Intrinsic), but if you look closely, you will notice that we invert the X and Z values. What I mean by that, is that we use Gamma (Z) as our X-axis in our pose and Alpha (X) as our Z axis pose
https://github.com/Kinovarobotics/ros_kortex/blob/5bc9ba69dff91707c4700e701433310589d21356/kortex_driver/src/non-generated/driver/kortex_arm_simulation.cpp#L252-L258

I know all of this is very confusing (especially when I make mistakes and swap Intrinsic and Extrinsic), but the documentation is correct and angles are computed following the z-y-x Tait-Bryan extrinsic convention. 

I hope this clarifies a bit

Best,
Felix

Tirine, 2022-03-23:
> We start at [0,0,0] [image]
> 
> we do a 45 degrees rotation on the Z-axis [image]
> 
> then we do a 90 degrees rotation on the X-axis [image]
> 
> Which brings the Z-axis alongs the diagonal 45 degree of the robots X and -Y plane.

But this is an Intrinsic rotation... 
As [this is](https://upload.wikimedia.org/wikipedia/commons/8/85/Euler2a.gif), even though this is in zxz intrinsic rotation.

But first you do a Z rotation, then ofc 0, so doesn't matter, but then you do the 90 deg X rotation on the NEW, Intrinsic, reference frame of the TCP, not the original fixed, extrinsic, motionless, world/robot/base reference frame, which is obvious as the X axis remains fixed even though it's ay 45deg from the world frame X axis

Tirine, 2022-03-23:
> ROS Kortex does use `GetEulerZYX` (which is ZYX Intrinsic), but if you look closely, you will notice that we invert the X and Z values. What I mean by that, is that we use Gamma (Z) as our X-axis in our pose and Alpha (X) as our Z axis pose

From the documentation again:
> First rotate around Z with alfa, then around the new Y with beta, then around new X with gamma.

So you are using it as Intrinsic ZYX, as it returns the values as also as ZYX, and you put it back in your pose as you say Alfa in Z and Gamma in X, but that is the correct for using it as an Intrinsic convention...

Tirine, 2022-03-25:
So basically, unless in Euler conventions, Extrinsic means internal, and Intrinsic means external, and it's written wrong in Wiki and every other source I've found about it (Or I've just completely lost my mind), I hope I've been able to convince you.

And yes, I tried to work with a Kinova Gen3 robot for, lets say, some embarrassing amount of months, miserably failing, as I continuously applied ZYX Extrinsic pose transformations (Except for in one place, where I used a library I thought was Extrinsic, but I noticed now as I go over it, it was Intrinsic all along.). Now I changed it to XYZ Extrinsic (or well, ZYX Intrinsic) and it's all just all of a sudden working.

And actually (just as a bonus) I can also, just by throwing around your images, show you how actual Extrinsic rotations would be step by step:

ZYX Extrinsic:
We start the same, at [0,0,0] [image]

We do a 45 degrees rotation on the Z-axis (Still the same as an intrinsic first step, as the world frame and object frame started the same.) [image]

Then comes the kicker!
We rotate the object, by 90 degrees in the World frame's X axis (Gotta say it's a bit hard to see the X Axis, but yeah, towards the view):
[image]
Again, same as the result from a XYZ Intrinsic.

And ofc we can do the same for XYZ Extrinsic:
We jump the start position, as it can be seen above.
We do a 90 deg rotation on the world X axis, and again, same as above, same as the Intrinsic first step, as the two reference frames started the same, there is no difference as the first rotation in a XYZ Intrinsic:
[image]

But then again comes the kicker!
We rotate the object, by 45 deg in the world frame's Z Axis:
[image]
Aaaand bam, we get the same position as from a ZYX Intrinsic, which is what you explained above.

Big difference being that the last rotation around a specific axis changes the direction of the objects axis (e.g. for XYZ Extrinsic rotation, when we apply the Z rotation, the object frame, intrinsic Z axis changes direction, as it's not parallel to the world frames, extrinsic Z axis)

felixmaisonneuve, 2022-04-04:
Hi @Tirine

Sorry for the delay, I have talked with a resource that worked on the development of this. 
Here is his answer. 

He first confirms that the robot uses the formula: Rz(theta_z) * Ry(theta_y) * Rx(theta_x).  
For a fixed referential (Extrinsic), rotations are applied on the x,y,z axes of the base in this order : X then Y then Z. 
For a mobile referential (Intrinsic), rotations are applied on the x,y,z axes of the mobile frame in this order : Z then Y' then X''

The question is therefore to know how this convention is named in the literature. Because it seems to vary from one source to another. According to Modern Robotics**:
This is the intrinsic ZYX convention. (Note that the term "Euler angle" normally refers only to the intrinsic convention, they are also sometimes called "Proper Euler Angles").
This convention is equivalent to the extrinsic XYZ convention.
It therefore seems that, indeed, the convention that we name in the documentation is not the correct one. 

** "Modern Robotics"  (Lynch, K. M., & Park, F. C. (2017). Modern robotics. Cambridge University Press.)
He tried to clarify what is going on based on references to properly define the nomenclature with regard to Euler angles. The clearest he has found to explain our problem is in Appendix B of Modern Robotics

Documentation (User guide and github repos) will be updated to reflect the Modern Robotics convention.

Please leave this issue open, I will update the repo and close it once it is done unless you have another question.

Thank you,
Felix