Kinovarobotics/Kinova-kortex2_Gen3_G3L #102: blending_radius in traj. point

im-renpei, 2022-03-25:
Hello!

When I customize a trajectory, the following 7 meta-data is required to form each trajectory point, what does "blending radius" mean?

```
def populateCartesianCoordinate(waypointInformation):
    
    waypoint = Base_pb2.CartesianWaypoint()  
    waypoint.pose.x = waypointInformation[0]
    waypoint.pose.y = waypointInformation[1]
    waypoint.pose.z = waypointInformation[2]
    waypoint.blending_radius = waypointInformation[3]
    waypoint.pose.theta_x = waypointInformation[4]
    waypoint.pose.theta_y = waypointInformation[5]
    waypoint.pose.theta_z = waypointInformation[6] 
    waypoint.reference_frame = Base_pb2.CARTESIAN_REFERENCE_FRAME_BASE
    
    return waypoint
```

felixmaisonneuve, 2022-03-25:
Hi @im-renpei 

The `blending_radius` (mm) is the distance from a given waypoint at which the robot starts moving towards the following waypoint.
[image]

This is to avoid going to a full stop at each waypoint. It makes the trajectory smoother. 
When you create your [WaypointList](https://github.com/Kinovarobotics/kortex/blob/master/api_cpp/doc/markdown/messages/Base/WaypointList.md#), you can specify `use_optimal_blending = true` and the arm will compute the optimal blending on its own so the trajectory is the quickest (in time). In this case, `blending_radius`, in your waypoints, will be ignored. 
You can also choose the use your own `blending_radius` if you specify `use_optimal_blending = false`. In which case, the `blending_radius` needs to be 0 for your first and last waypoint. 

Regards,
Felix

closed 2022-03-26 by im-renpei

im-renpei, 2022-09-20:
Hi Felix,

Hope you everything is well. I wonder if it is possible for you to share
the trajectory planning code with blend radius or where I can get the code
for reference?

Looking forward to hearing from you.
Pei