Kinovarobotics/Kinova-kortex2_Gen3_G3L #103: EE acceleration (low-level control)

JDchen-Jaden, 2022-04-04:
Hi!

How to get the acceleration of the end-effector using: `imu_acceleration_x` under low-level control? For example, this command is used to get information of actuators: `feedback.actuators(0).position()`

Thanks!

Best regards,
JD

felixmaisonneuve, 2022-04-04:
Hi @JDchen-Jaden,

You cannot get the acceleration of the end-effector in low level control. This information is computed by the base using Inverse (or forward, I can't remember) kinematics. When in Low level, the base is not computing anything and your are almost communicating directly with each actuators, so you won't get any BaseFeedback. If you want to know the acceleration of the end-effector, you would need to compute the inverse kinematics on your own and integrate the position to get the acceleration. 

Regards,
Felix

JDchen-Jaden, 2022-04-04:
Hi @felixmaisonneuve ,

Thanks a lot for your quick reply. 

I learned from the User Manual of Kinova Gen3 that the interface (assuming no end effector) has an accelerometer there. If this is the case, then the acceleration in the workspace should be measured directly. How the measured acceleration can be accessed? Thanks!

[image]

Best regards,
JD

felixmaisonneuve, 2022-04-04:
You are right, I thought `imu_acceleration_x` was received from the [BaseFeedback](https://github.com/Kinovarobotics/kortex/blob/master/api_cpp/doc/markdown/messages/BaseCyclic/BaseFeedback.md#), but it is actually coming from the interconnect [Feedback](https://github.com/Kinovarobotics/kortex/blob/master/api_cpp/doc/markdown/messages/InterconnectCyclic/Feedback.md#).

In that case, you should be able to access it using `feedback.interconnect.imu_acceleration_x()`

JDchen-Jaden, 2022-04-04:
Sorry for the misleading question. I'll give it a try. Thanks a lot! 

By the way, I am a bit confused about how to use the message documentation. I can find certain methods associated with actuators, base, and interconnect, for example  `imu_acceleration_x()` in "_/markdown/messages/InterconnectCyclic/Feedback.md_", but I have no idea how to use this method since I have trouble to find any documentation that introduces the syntax of `interconnect` or `actuators`  before the method been called. It would be helpful if there is any document there. Thanks! 

Best regards,
JD

felixmaisonneuve, 2022-04-04:
About the documentation, what I usually do is look at the [Summary Pages](https://github.com/Kinovarobotics/kortex/tree/master/api_cpp/doc/markdown/summary_pages). From there, you can search for the "module" (base, base_cyclic, actuator_config, vision...) you are using. 
In your case, you are using the `Refresh` or `RefreshFeedback` function from the Base_Cylic module, so you you look inside _summary_pages/BaseCyclic.md_. In there, you can see all the available functions. In there you will find `Refresh` and `RefreshFeedback` with the output type, input type and the description. You see that `Refresh` returns a `Feedback` object. This is also a link that you can follow to get all the members and functions for this object. From there you see there is feedback coming from the base, the actuators and the interconnect. You can go down all you need

closed 2022-04-05 by JDchen-Jaden