Kinovarobotics/Kinova-kortex2_Gen3_G3L #104: Torque control

JDchen-Jaden, 2022-04-08:
Hi,

I am working on implementing torque control based on the user interaction force (at the endpoint) of Kinova Gen3. 

The following is what I tried to send the torque command. However, the robot tends to move back to the initial position once the user's exerted force is released every time. I've tried: (1) adding the initial torque to the torque command; (2) setting the actuator torque offset to zero, But neither works.  Can anyone help? Thanks!
```
                for(int i = 0; i < 7; i++)
                {
                    if(i==0 || i==3 || i ==5){
                        base_command.mutable_actuators(i)->set_position(feedback_.actuators(i).position());
                        base_command.mutable_actuators(i)->set_torque_joint(torque_command[i]);
                      // base_command.mutable_actuators(i)->set_torque_joint(init_first_torque[i] + torque_command[i]);
                    }
                }
```

felixmaisonneuve, 2022-04-08:
Hi @JDchen-Jaden,

Is your `feedback_` updated correctly?

Based on what I see, I feel like the position command is not updated, so the arm will move a bit (maybe 2-3cm), then it will reach the position threshold and the actuator will promptly go back to the initial position

Does that describe what you are seing?

Best,
Felix

JDchen-Jaden, 2022-04-08:
Hi @felixmaisonneuve 

`feedback_` has been set as a global variable for updating in a callback function. It should be updated correctly. 
```

k_api::BaseCyclic::Feedback feedback_;

void my_func(const k_api::BaseCyclic::Feedback feedback){
    feedback_ = feedback;
}

void my_callback(const k_api::Error& err,const k_api::BaseCyclic::Feedback& feedback)
{
    my_func(feedback);
}

int main(){
   feedback_ = base_cyclic->RefreshFeedback();
// Send a first frame
   feedback_ = base_cyclic->Refresh(base_command);

       loop{
           base_command.mutable_actuators(i)->set_position(feedback_.actuators(i).position());
           base_command.mutable_actuators(i)->set_torque_joint(torque_command[i]);
       }
}
```
According to the torque control example and the generalized coordinates given in the user manual, the torque reading of **joint 1** needs to flip the sign. When I flip the sign of computed torque command sending to **joint 1** it rotates fast which is incorrect. Do I need also to flip the sign?  

For setting actuator torque offset to be zero, first, the robot moves to the zero-configuration, when I click the "SET OFFSET" button, it is still in grey. I am not sure if it has been set successfully or not.  Thanks!
[image]

closed 2022-04-11 by JDchen-Jaden