Kinovarobotics/Kinova-kortex2_Gen3_G3L #105: RefreshCommand method return  unsuported method.

btabia, 2022-04-11:
## Description

One paragraph to explain the issue

## Version

At a minimum, provide the Kortex API and Kortex-enabled device versions.

KortexAPI :

Kortex Device :

## Steps to reproduce

When calling the method base_cyclic->refreshCommand(base_command,0) ; the system return the following message: 

Error sub-code: UNSUPPORTED_METHOD
Kortex exception: Server protocol error, Error sub type=UNSUPPORTED_METHOD => <srv: 3, fct: 2, msgType: 3>

## Code example

void KinovaRobot::sendCommand()
{
  auto frame_id = base_command->frame_id() + 1;
  base_command->set_frame_id(frame_id);
  if (frame_id > 65535) base_command->set_frame_id(0);

  Eigen7f position = get_pose();

  auto torque_offset = getTorqueOffset();
  auto torque_command = getCommand();

  for (unsigned int i = 0; i < KinovaRobot::actuator_count; i++) 
  {
    base_command->mutable_actuators(i)->set_command_id(frame_id);
    base_command->mutable_actuators(i)->set_position(position(i));
    base_command->mutable_actuators(i)->set_torque_joint(torque_command(i));
  }

  base_cyclic->RefreshCommand(*base_command,0);

}

btabia, 2022-04-12:
Both my firmware and my the kortex api are in version 2.3 

[image]

felixmaisonneuve, 2022-04-12:
Ok, I think this is because you have a typo in your function so the arm does not recognise the command your are sending
try `RefreshCommand` instead of `refreshCommand`

Best,
Felix

felixmaisonneuve, 2022-06-08:
Hi @btabia,

I will assume this has solved the problem and close this issue since I didn't ear from you.
If you still have a problem, you can re-open it or leave a comment and I will be glad to assist you.

Best,
Felix

closed 2022-06-08 by felixmaisonneuve