Kinovarobotics/Kinova-kortex2_Gen3_G3L #106: Robot returning Wrong servoing mode

btabia, 2022-04-11:
## Description

When using the base_cyclic->refresh(base_command,0), the robot return the following message: 

Kortex exception: Device error, Error sub type=WRONG_SERVOING_MODE => <srv: 3, fct: 1, msgType: 3>
description: Wrong servoing mode, must be low level servoing mode
Error sub-code: WRONG_SERVOING_MODE

void KinovaRobot::sendCommand()
{
  auto frame_id = base_command->frame_id() + 1;
  base_command->set_frame_id(frame_id);
  if (frame_id > 65535) base_command->set_frame_id(0);

  Eigen7f position = get_pose();

  auto torque_offset = getTorqueOffset();
  auto torque_command = getCommand();

  for (unsigned int i = 0; i < KinovaRobot::actuator_count; i++) 
  {
    base_command->mutable_actuators(i)->set_command_id(frame_id);
    base_command->mutable_actuators(i)->set_position(position(i));
    base_command->mutable_actuators(i)->set_torque_joint(torque_command(i));
  }

    /*auto lambda_fct_callback =
      [this](const Kinova::Api::Error &err,
             const Kinova::Api::BaseCyclic::Feedback data) {
        // Set current pose, and velocity
        this->getFeedback(&data);
      };

  base_cyclic->Refresh_callback(*base_command,lambda_fct_callback, 0);*/

  base_cyclic->Refresh(*base_command,0);

}

felixmaisonneuve, 2022-04-11:
Hi @btabia,

The error message indicates you are not in LOW_LEVEL control.
Can you make sure you put your base in LOW_LEVEL control mode before calling this function : 
https://github.com/Kinovarobotics/kortex/blob/0aca9b23b0f04d76bf27e3f7b45befc2c77ebed9/api_cpp/examples/108-Gen3_torque_control/01-torque_control_cyclic.cpp#L219-L220

Best,
Felix

btabia, 2022-04-12:
Hello, 

I have added the line that you mentioned about the servoing mode but now I have this kind of error appearing when running my control system. 

[libprotobuf FATAL /home//.conan/data/kortex_api_cpp/2.3.0-r.34/kortex/stable/package/c023db9fc677d4d0b3bd0c20f71385e4cf8a1220/thirdParty/include/google/protobuf/repeated_field.h:1522] CHECK failed: (index) < (current_size_): 
Unknown error.

This is an error with Protobuf, how the error could be resolved?

Many thanks

felixmaisonneuve, 2022-04-12:
Yes, this happens when you try to access an element out of bounds in a protobuf repeateable field. 
The error is coming from your code. You are trying to access an element outside of range.

Try to figure out which command has an index out of bounds. I don't know what you are trying to do, but usually it's a good idea to start from one of our example that we know work and doing small changes. It becomes easier to find out what is not working instead of writing a whole program from the beginning and trying to resolve everything at once. 

Best,
Felix

felixmaisonneuve, 2022-06-08:
Hi @btabia,

I will close this issue as well. If you weren't able to find the source of the error however, you can always re-open the issue.

Best,
Felix

closed 2022-06-08 by felixmaisonneuve