Kinovarobotics/Kinova-kortex2_Gen3_G3L #108: Energy of Arm

RobaczeQ, 2022-04-19:
## Description

Energy in arm doesn't add up. 

## Version

KortexAPI : 2.3

Kortex Device : Kinova gen3 7 DoF

## Steps to reproduce

It's possible to access Voltage and Current from Website/Matlab/Python/C++, so:
1. choose one method (from above)
2. move robot arm to home position
3. read `BaseFeedback` `arm_voltage` `arm_current` and multiply them. 
4. read `ActuatorFeedback` `voltage` `current_motor` of each arm, multiply them and add them together. 

## Expected behavior

Value from `BaseFeedback` is ~24 W, measuring Wattage from outlet is about ~34 W, which should mean ~0.70 efficiency. From detailed information about power adapter, efficiency should be ~0.88. Value from `ActuatorFeedback` is ~65 W.

1. Is `BaseFeedback` main voltage/current of whole robot or I should add some additional part from somewhere else (like gripper)?
2. How can value added up from each joint (`ActuatorFeedback`) be higher than power from outlet?

felixmaisonneuve, 2022-04-25:
Hi @RobaczeQ,

For point 1, I would expect the value from `BaseFeedback` to be the full arm current at 24V. If I understand correctly you are expecting a highier number than 24W? According to your power adapter, you should expect something like 30W (0.88 * 34)? Or is 0.88 the efficiency of your power adapter?

For point 2, I talked with a resource and he told me the current in the actuator feedback is the motor current, not the current consumed by the actuator on the 24V bus. The motor is controlled by Field Oriented Control (FOC). It is apparently very common. The voltage to the motor is modulated (so not 24V) to obtain a certain current. So the power consumed by the motor is not equal to "motor current x 24V" since the motor is not at 24V. The voltage to the motor is not given in the feedback, since it is the current which gives information on the stress of the hardware components of the actuator and on the torque generated by the motor.
Does that make sense to you?

Best regards,
Felix

RobaczeQ, 2022-04-25:
Hi @felixmaisonneuve 

1. I use power adapter from Kinova package (TDK Lambda AC Power adapter DTM300PW240D2). I expected 30W (0.88 * 34), but if you say that's current of full arm, it's ok. Just wanted to know if I can use that to measure how much energy Kinova uses. 
2. So `voltage` from `ActuatorFeedback` is 24V bus and `current_motor` is i_q current from FOC (i_d is assumed to be 0?). Torque could be computed from that current if torque constants of motors would be known, right? 

Best regards,
Matt

closed 2022-04-28 by RobaczeQ