Kinovarobotics/Kinova-kortex2_Gen3_G3L #111: High-level gripper command example: "Failed to execute gripper command: Unimplemented"

jimmyyhwu, 2022-05-31:
## Description

Running the high-level gripper command example (106-Gripper_command/01-gripper_command.py) gives the following error:

kortex_api.Exceptions.KServerException.KServerException: Server error name=ERROR_DEVICE, sub name=UNIMPLEMENTED => Failed to execute gripper command: Unimplemented 

## Version

KortexAPI: kortex_api-2.3.0.post34-py3-none-any.whl
Kortex Device: Gen3-2.3.0.swu (firmware bundle 2.3.0-r.227)

## Steps to reproduce

Run the example script: https://github.com/Kinovarobotics/kortex/blob/master/api_python/examples/106-Gripper_command/01-gripper_command.py

## Any other information

We have multiple arms, and I have tried different combinations of base/arm/gripper. All arms use kortex_api-2.3.0 and firmware bundle 2.3.0-r.227. This error only seems to occur with one arm in particular, no matter what base or gripper it is attached to. All combinations I have tried seem to work as expected, as long as they don't include this arm.

Note that for the problematic arm, the low-level gripper example (107-Gripper_low_level_command/01-gripper_low_level_command.py) works as expected, it's just the high-level example that gives an error.

felixmaisonneuve, 2022-06-02:
Hi @jimmyyhwu,

Is the gripper correctly configured with the gripper?
[image]

Have you tried to factory reset the arm? 

Best,
Felix

jimmyyhwu, 2022-06-03:
I configured the gripper based on your attached screenshot and it works now, thanks!

closed 2022-06-03 by jimmyyhwu