Kinovarobotics/Kinova-kortex2_Gen3_G3L #117: timeout error

JDchen-Jaden, 2022-07-21:
Hi!

My Gen3 always has an error recently when conducting low-level torque control: `Error: timeout detected: BaseCyclicClient::Refresh
`

Resetting the robot solves the problem temperately. More specifically, after presenting it can run the program only one time. Running the same program the second time, the above error occurs. Meanwhile, the robot does not respond, and the program is terminated. **This issue has only occurred recently**. 

Any suggestion would be appreciated. 

Thanks,
JD

JDchen-Jaden, 2022-07-21:
This might be a hardware issue since the red LED is on. I may contact Kinova support for further help. But, still, your suggestion would be appreciated.

felixmaisonneuve, 2022-07-21:
Hi @JDchen-Jaden,

A red LED is not very good. You probably have an error message in the web app that could help identify the issue?
Based on the error in the web app, we may be able to determine if this issue can be solved on the software side or if it is a hardware issue. 

Best,
Felix

JDchen-Jaden, 2022-07-21:
Thanks @felixmaisonneuve 

The problem is that the WebApp can not be accessed (Connection Error. Make sure Robot is started) though the robot was on and the Ethernet cable was connected.

felixmaisonneuve, 2022-07-22:
In this case it would be better to contact the support team at kinova@support.ca. There is not much I can do. There seems to be a major issue with your arm since the LED is red and you cannot connect to the webapp. 

The red LED and the timeout issue might or might not be related, but you should definitely try to fix it before trying to diagnose your low-level control.

If our support team is able to "repair" your arm, but you still get the timeout error, leave a comment here and I will help you on that. You can leave this issue opened for now.

Regards,
Felix