Kinovarobotics/Kinova-kortex2_Gen3_G3L #119: Kortex API was not downloaded prior to running CMake.

brianngoh, 2022-07-25:
## Description

When I try to follow the instructions to get the kortex running on C++ there is an error that says. Kortex API was not downloaded prior to running CMake. CMake Error at CMakeLists.txt:187 (message):. I followed the instructions listed on https://github.com/Kinovarobotics/kortex/blob/master/api_cpp/examples/readme.md. I also followed the youtube tutorial on https://www.youtube.com/watch?v=JparQ69LMzY&list=PLz1XwEYRuku5OteYpOevN6VV_agK5G7zu&index=3 and still had the same error. When I downloaded the kortex api for c++ and transferred the files over to the kortex_api in the kortex folder, I ran it through the terminal and it was not able to build the folder. 

## Version

KortexAPI :
This issue is for kortex api 2.3

Kortex Device :
Kinova gen 3 robot. 

## Steps to reproduce

1. Download Conan  

2. build code in the examples directory 

3. Run the code from a network or wired connection 

All instructions on https://github.com/Kinovarobotics/kortex/blob/master/api_cpp/examples/readme.md

## Code example
This is all in the example directory of the api_cpp folder which is located in kortex. I am using the terminal to put in these commands. 

mkdir build  

cd build  

cmake .. -DCMAKE_BUILD_TYPE=release -DOWNLOAD_API=OFF -DKORTEX_SUB_DIR=/home/brianngoh/kortex/api_cpp/examples/kortex_api

make  

## Expected behavior

Kortex API was not downloaded prior to running CMake.
CMake Error at CMakeLists.txt:187 (message):

-- Configuring incomplete, errors occurred!
See also "/home/brianngoh/kortex/api_cpp/examples/CMakeFiles/CMakeOutput.log".

## Any other information

I have been working on trying to figure this out for several days and it would be extremely helpful if someone can help. I downloaded kortex but for some reason the computer does not recognize the kortex_api file which I updated with the correct kortex api documents.

felixmaisonneuve, 2022-07-27:
Hi @brianngoh,

You indicate that you downloaded Conan, but the commands you used and the error message tell me you are trying to build without using Conan. 
There is two way you can use to build our examples : by using Conan and by building manually. Both should work and are very valid choices.
Building with conan should be easier, but building manually should not be too hard either. You seem to have mixed both method and it may explain why you are having issues. 
With Conan, you should only have to do (if it is installed and correctly setup, which should be your case) : 
```
cd /home/brianngoh/kortex/api_cpp/examples
./scripts/build-gcc.sh release
```
You don't have to worry about downloading the API with conan, it should do it automatically. Just make sure the `USE_CONAN` variable is correctly set to `ON`
https://github.com/Kinovarobotics/kortex/blob/2218dd6bb9fa2d135f77d1f611ffe8d3cce2edf9/api_cpp/examples/CMakeLists.txt#L43
if it doesn't work, can you print the full output and your conan profile info (`conan profile show default`)?

If you would rather build manually, I can help you set up as well. 

Regards,
Felix

felixmaisonneuve, 2022-08-11:
Hi @brianngoh,

Do you still need assistance with this issue or can we close it?
I will close it anyway in a couple days if I do not ear from you

Best,
Felix

brianngoh, 2022-08-11:
Sorry for the late reply you can close it.

closed 2022-08-11 by felixmaisonneuve