Kinovarobotics/Kinova-kortex2_Gen3_G3L #120: When are protection zones active?

etaoxing, 2022-08-05:
I am trying to do continuous control (Cartesian position) with the Kinova Gen3, so goal pose / trajectory is not known beforehand.

I know [protection zones](https://github.com/Kinovarobotics/kortex/blob/master/api_python/examples/100-Overview/02-protection_zones_configuration.py) are active for [cartesian](https://github.com/Kinovarobotics/kortex/blob/2218dd6bb9fa2d135f77d1f611ffe8d3cce2edf9/api_python/examples/102-Movement_high_level/01-move_angular_and_cartesian.py#L115) but not [angular](https://github.com/Kinovarobotics/kortex/blob/2218dd6bb9fa2d135f77d1f611ffe8d3cce2edf9/api_python/examples/102-Movement_high_level/01-move_angular_and_cartesian.py#L80) actions.

If I use [twist commands](https://github.com/Kinovarobotics/kortex/blob/2218dd6bb9fa2d135f77d1f611ffe8d3cce2edf9/api_python/examples/102-Movement_high_level/03-twist_command.py#L86), are protection zones active? This [function](https://github.com/Kinovarobotics/kortex/blob/2218dd6bb9fa2d135f77d1f611ffe8d3cce2edf9/api_python/examples/100-Overview/02-protection_zones_configuration.py#L84) seems to imply yes.

felixmaisonneuve, 2022-08-05:
Hi @etaoxing 

Yes protection zones should be active for Twist commands as well. Is this not the case for you?

Best,
Felix

etaoxing, 2022-08-08:
Seems to work, thanks for the clarification!

closed 2022-08-08 by etaoxing