Kinovarobotics/Kinova-kortex2_Gen3_G3L #121: Conversion error trying to replicate Kinova youtube example

RobaczeQ, 2022-08-30:
## Description

Code from Kinova youtube tutorial (https://youtu.be/5PB--gaI6rI?t=1023) doesn't compile on Linux, but it compiles on Windows using Visual Studio Compiler. There were the same questions on the video from 10 months ago, but no replay. 

## Version

KortexAPI : 2.3

Kortex Device : Kinova gen3 7 DoF

## Steps to reproduce

Get example from video (https://youtu.be/5PB--gaI6rI?t=1023)
That line creates an error:
std::function<void(Kinova::Api::Base::ActionNotification)> actionCallback = std::bind(&KortexRobot::OnActionNotificationCallback, this, std::placeholders::_1);

Error:
conversion from ‘std::_Bind_helper<false, void (KortexRobot::*)(Kinova::Api::Base::ActionNotification&), KortexRobot*, const std::_Placeholder<1>&>::type’ {aka ‘std::_Bind<void (KortexRobot::*(KortexRobot*, std::_Placeholder<1>))(Kinova::Api::Base::ActionNotification&)>’} to non-scalar type ‘std::function<void(Kinova::Api::Base::ActionNotification)>’ requested

## Expected behavior

I would like to have the same code for Linux and Windows, is it possible to change it so it works without Visual Studio Compiler, which isn't available on Linux?

felixmaisonneuve, 2022-08-30:
Hi @RobaczeQ,

What GitHub example is this from? Youtube examples are ususally taken from our github repo.
Looking at your error message it looks like you have a function like this :  
```
void OnActionNotificationCallback(Kinova::Api::Base::ActionNotification& actionNotification)
```
and it cannot convert the `Kinova::Api::Base::ActionNotification&` to `Kinova::Api::Base::ActionNotification`. Maybe try to change your `OnActionNotificationCallback` definition to something like this : 
```
void OnActionNotificationCallback(Kinova::Api::Base::ActionNotification actionNotification)
```

Best,
Felix

RobaczeQ, 2022-08-30:
I was also wondering why YouTube examples aren't uploaded to Github. Some of the videos are just GIthub examples, but some don't. 
But you're right, now it's working on Linux, strange that did compile on Windows. 

Thank you for your help and time,
Matt