Kinovarobotics/Kinova-kortex2_Gen3_G3L #125: External force on the Gripper of the gen3

wangxRobot, 2022-09-14:
## Summary

Is there a way or interface to capture the external force applied to the Gripper of the gen3？Do we need to fit a force sensor to the end-effector？

## Use case

We need to use gen3 for the task of impedance control, which requires us to obtain the external forces.

## Alternative

I know that the external force can be calculated by joint torque, but this requires calculating moment of gravity and moment of inertia, and the claws need to grasp things, which leads to the complexity of the calculations.

felixmaisonneuve, 2022-09-14:
Hi @wangxRobot,

In the [BaseFeedback](https://github.com/Kinovarobotics/kortex/blob/master/api_cpp/doc/markdown/messages/BaseCyclic/BaseFeedback.md#), you have access to `tool_external_wrench_force_x/y/z` and `tool_external_wrench_torque_x/y/z` that are the external forces and torques applied on the gripper. This is computed from all joints torques (the alternative you mentionned). Because of this, it might not be super accurate (resetting the torque offset for all joints might help a bit) and this is mostly for external forces when the arm is in a static position. Accuracy will probably be a lot worse if the arm is moving (I did not try it). Also, this feedback is only available in High Level control. 
If you want highier accuracy, you might need to fit a force sensor to the end-effector.

For the grasping force, this is not supported, meaning you cannot know, while closing the gripper what force is applied by the gripper. If thats something you need, there is a workaround where you monitor the position of the gripper and stop the movement once the gripper is not closing anymore, but it is not accurate at all.

I will link these other issue that were about this topic : #44, #63 and #87.

Best Regards,
Felix

felixmaisonneuve, 2022-11-16:
Hi @wangxRobot,

Did you try it? Did it work?
I will close this issue in a couple days unless you require more support, but I don't know what you could do if these options do not work.

Best,
Felix

closed 2022-12-15 by felixmaisonneuve