Kinovarobotics/Kinova-kortex2_Gen3_G3L #126: Read a snapshot JSON pose or sequence

rduke199, 2022-09-15:
What is the best way to read a snapshot JSON file (either a pose snapshot or a sequence) and translate that to robotic action? Do the `Base_pb2.Action()` and `Base_pb2.Sequence()` classes have methods to read in a JSON file?

felixmaisonneuve, 2022-09-15:
Hi @rduke199,

Unfortunately, there is nothing built-in in the arm that allows to convert from JSON to Action and from Action to JSON. 
You will need to implement your own parser to read a JSON file and create the action according to the JSON file. 

This shouldn't be to complicated to implement, but yeah, you will need to do it yourself

Best,
Felix

felixmaisonneuve, 2022-09-16:
A coworker told me about this protobuf tool : https://googleapis.dev/python/protobuf/latest/google/protobuf/json_format.html

I have never used it, but he told me it works great. With the right arguments, it will even be recursive and convert the subtypes.  The old version of protobuf bundled with the gen3 API has some quirks at the Enums level if you want to re-serialize to JSON (it falsely specifies `use_integers_for_enums`) but otherwise it works fine according to him.

Best,
Felix

rduke199, 2022-09-16:
Thank you. How would you convert the `protobuf` class to a `Base_pb2.Sequence()` class?

felixmaisonneuve, 2022-09-16:
`Base_pb2.Sequence()` is a protobuf object already

felixmaisonneuve, 2022-11-16:
Hi @rduke199,

Does the protobuf tool work for you? 
Do you require more assisstance?
I will assume it works and close this issue in a couple days if I don't hear from you.

Best,
Felix

rduke199, 2022-11-16:
I don't know if that tool works because I developed another work around.

felixmaisonneuve, 2022-11-16:
Great, whatever works for you
Closing this issue then

Best Regards,
Felix

closed 2022-11-16 by felixmaisonneuve