Kinovarobotics/Kinova-kortex2_Gen3_G3L #127: Continuous Joint Control in While Loop

JaeYoungLee79, 2022-10-13:
## Summary
Operation flow is : 
Input value of end-tip position --> Inverse Kinematics --> Calculated Target Joint Value --> Joint Control as continuously 

## Status
I'm coding above the flow and It works as well based on "[01-move_angular_and_cartesian.py]".
Problem is that once running.
I have four XYZ position sets. it is rectangular points.

I already control  the position of end-tip in Cartesian space by using "03-twist_command.py" example.
Based on "While Loop", a change amount of each X,Y,Z inputs to Twist Command.
It woking well to tracking the pixel value of video capture as my another application coding.

In case of Joint Level, the example of "01-move_angular_and_cartesian.py"
-------------------------------------------------------------------------------------------
# Maximum allowed waiting time during actions (in seconds)
TIMEOUT_DURATION = 100
~~~~~
#Calculated Target joint value of First Position 
#And Joint value input to as below
~~~~~~~
joint_angle = action.reach_joint_angles.joint_angles.joint_angles.add()
~~~~~
print("Executing action")
base.ExecuteAction(action)
# It wokring well. And Second joint value input to as below
~~~~~~~
joint_angle = action.reach_joint_angles.joint_angles.joint_angles.add()
~~~~~
print("Executing action")
base.ExecuteAction(action)
>>> It is not working.
------------------------------------------------------------------------
It connection time is enough to operation as my just thinking.
Is possible that continuous joint value control in real-time? (as like the twist command example) 
The example of waypoint trajectory already has the joint value of each point. 
Each loop step --> calculate Joint values --> Throw value for Joint control command

Thanks for reading.
Best Regards

JaeYoungLee79, 2022-10-13:
## Question
It connection time is enough to operation as my just thinking.
Is possible that continuous joint value control in real-time? (as like the twist command example) 
The example of waypoint trajectory already has the joint value of each point. 
Each loop step --> calculate Joint values --> Throw value for Joint control command

Thanks for reading.
Best Regards

JaeYoungLee79, 2022-10-17:
As I look insdie API manual
Below It can operate at "While Loop" API
- ChangeJointSpeeds.md
- ChangeTwist.md
- ChangeWrench.md

I can't looking "Angle Control"
I want to control changes of angle value.

Anybody this experiences?

Thanks
Best Regards

felixmaisonneuve, 2022-10-19:
Hi @JaeYoungLee79,

I am not sure I fully understand what you want to do, so I will explain why what you are doing is not working and hopefully it will help you.

First, when you create your `reach_joint_angle` with your and create the `joint_angles` in a loop like this
```py
    for joint_id in range(actuator_count.count):
        joint_angle = action.reach_joint_angles.joint_angles.joint_angles.add()
        joint_angle.joint_identifier = joint_id
        joint_angle.value = 0
```
Your `action.reach_joint_angles.joint_angles.joint_angles` is initially empty, so you need to add your joint angles with `.add()`. In you loop, it will look something like this (I will assume a 7DOF arm, so `range(actuator_count.count)` = [0..7])
```py
action.reach_joint_angles.joint_angles.joint_angles = [] # initially, then you add {joint_id, 0} in each loop
action.reach_joint_angles.joint_angles.joint_angles = [{0, 0}] # after the first loop
action.reach_joint_angles.joint_angles.joint_angles = [{0, 0}, {1, 0}] # after the second loop
action.reach_joint_angles.joint_angles.joint_angles = [{0, 0}, {1, 0}, {2, 0}] # after the third loop
...
action.reach_joint_angles.joint_angles.joint_angles = [{0, 0}, {1, 0}, {2, 0}, {3, 0}, {4, 0}, {5, 0}, {6, 0}] # after the whole for loop
```

You finish with 7 `JointAngle` in your `action.reach_joint_angles.joint_angles.joint_angles`. If you `.add` another `JointAngle`, you would end up with more `JointAngle` in your action then the number of joints on your arm and it won't work. 
If you want to send another `reach_joint_angles` Action using the same object you created, you will want to edit the current values with something like this 
```py
    for joint_id in range(actuator_count.count):
        joint_angle = action.reach_joint_angles.joint_angles.joint_angles[joint_id] # instead of .add
        # joint_angle.joint_identifier = joint_id # not needed since it is already at the correct ID
        joint_angle.value = 0 # you can change the value here
```

I could help you more if you sent maybe a small sample of your code so I could better understand what you are trying to do and why it is not working

Best,
Felix

JaeYoungLee79, 2022-10-26:
Thanks for answer!
It solved!

closed 2022-10-26 by JaeYoungLee79