Kinovarobotics/Kinova-kortex2_Gen3_G3L #132: Network Error during working

pouyan-asg, 2022-11-15:
Hi,
I am using Gen3, and I faced a network error while working. Before, it worked without a problem. I can control it through Web App, and I changed the Ethernet cable with a new one to make sure. 
I should tell the robot movement is close to each other in a range below 50 cm. As I mentioned before, there was no problem before, and only today I encountered some times.
Do you have any suggestions to figure it out? 
Thank you very much for your time and consideration.

felixmaisonneuve, 2022-11-16:
Hi @pouyan-asg,
Is your [utilities.py](https://github.com/Kinovarobotics/kortex/blob/master/api_python/examples/utilities.py) file in your python workspace?
What is the error message?

Best,
Felix

pouyan-asg, 2022-11-16:
Hi @felixmaisonneuve 

Yes, the "utilities" file is in my workspace, and this situation happened just yesterday for some time. Previously, I had no issue in this regard. 
I did not record the error, but I could see it's a "Network Error" in the Fault section. 
One question, if the robot moves through close points (less than 10 cm & Cartesian waypoints), is it a reason for faults (network or even joint)? I am asking this question because the errors happened when the robot wanted to move between very close points. This is only an opinion, and I am not sure there is a direct link between movements and errors.

Thank you very much

felixmaisonneuve, 2022-11-16:
There is no link between movements and network error that I know of. 
If you are not able to reproduce the error, there is no way to know if the error was coming from the arm, from the network hardware (cable/switch/router...), you computer network config or somewhere else. 
I will close this issue for now, but if you experience this issue again, please reopen it and leave some more details : Your computer IP, your arm IP, the error message in the API/WebApi and anything else that could be important.

Regards,
Felix

closed 2022-11-16 by felixmaisonneuve