Kinovarobotics/Kinova-kortex2_Gen3_G3L #137: get the external force applied on ee of gen3

Myfirsttrying, 2022-12-10:
## Summary

Hi, I have a question about how to read the end-effector force through API. According to the file in the api_python, I could read end-effector force through basecyclic -> 'tool_external_wrench_force_x/y/z'. But when i write a 'while' cycle by myself  to get the real-time external force applied on end-effector and 'print' it, it is just print a value and never change, though i put my hand on end-effector to apply different forces. i don't know why and request help.

## Use case

Why is this feature needed?

What are the use case covered by this feature?

## Alternative

Myfirsttrying, 2022-12-10:
by the way, it is in the high level control (SINGLE_MODE in scripts)

Myfirsttrying, 2022-12-10:
I think it only measures and outputs the value when the program starts running, not refreshes it in real time. Because when I re-run it, I can get another set of different values, but still unchanged. I don 't know where the problem is and how to solve it.

felixmaisonneuve, 2022-12-15:
Hi @Myfirsttrying,

Yes, like you said, it probably only measures and outputs the value when the program starts running, and values are not refreshed. 

This is probably an error in your code, can you share your code. your feedback value should always be refreshed in your while loop with with `feedback = base_cyclic.RefreshFeedback()`. If you never call `RefreshFeedback()`, your tool_wrench will never be updated either.

Best,
Felix

Myfirsttrying, 2022-12-16:
> Hi @Myfirsttrying,
>
> Yes, like you said, it probably only measures and outputs the value when the program starts running, and values are not refreshed.
>
> This is probably an error in your code, can [...]

Thank you very much for the kind answer, I will once again carefully check my code, if still unable to solve I will come again to seek your help. By the way, I know that GEN3 has admittance control mode, can I use API to switch it to this control mode in my code ? and how to implement that?

felixmaisonneuve, 2022-12-16:
Yes, you can change the admittance mode with the API. It will something like
```py
admittance = Base_pb2.Admittance()
admittance.admittance_mode = Base_pb2.JOINT
base.SetAdmittance(admittance)
```

I used python, but C++ code should be fairly identical. 
We use the `SetAdmittance` function from the [base](https://github.com/Kinovarobotics/kortex/blob/master/api_python/doc/markdown/summary_pages/Base.md). This function takes an [Admittance](https://github.com/Kinovarobotics/kortex/blob/master/api_python/doc/markdown/messages/Base/Admittance.md#) object in parameters. This object than has a `admittance_mode` coming from the [AdmittanceMode](https://github.com/Kinovarobotics/kortex/blob/master/api_python/doc/markdown/enums/Base/AdmittanceMode.md#) enum

felixmaisonneuve, 2022-12-16:
Since you want cartesian adittance, you would do 
```py
admittance.admittance_mode = Base_pb2.CARTESIAN
```

Please do not do something like this.
```py
control_mode_info = ActuatorConfig_pb2.ControlModeInformation()
control_mode_info.control_mode = 1
control_mode = ActuatorConfigClient(router)
control_mode.SetControlMode(control_mode_info)
```

closed 2023-02-22 by felixmaisonneuve