Kinovarobotics/Kinova-kortex2_Gen3_G3L #141: Camera extrinsic update

jabbershort, 2023-01-03:
## Description

The camera extrinsic value (captured using the example script '02-vision_extrinsics.py') does not update when the arm is moved (unless set manually).

## Version

At a minimum, provide the Kortex API and Kortex-enabled device versions.

KortexAPI : 2.5.0

Kortex Device : 2.3.0-r.227

## Steps to reproduce

1. Run example script to get extrinsic. (I removed the 'set' function of the extrinsics).
2. Move arm
3. Run script again, same result.

## Expected behavior
The VisionConfigClient.GetExtrinsicParameters() should give live camera frame position. (Or is there a way of forcing it to update itself?) I'm mostly interested in the color camera frame, I know I could use the joint angles and the transformation matrix to calculate it, but I was hoping there was a built in method to achieve this.

felixmaisonneuve, 2023-01-03:
Hi @jabbershort,

I am not super familiar with intrinsic/extrinsic parameters, but the [ExtrinsicParameters](https://github.com/Kinovarobotics/kortex/blob/master/api_cpp/doc/markdown/messages/VisionConfig/ExtrinsicParameters.md#) documentation page says : 
> The rotation matrix from depth sensor reference frame to color sensor reference frame

So, the way I understand it, the rotation and translation matrix are used to correlate depth and color images. Maybe I do not understand it correctly, but I don't se why this matrix should update on its own (e.g. when moving the arm). 

If you are looking for the camera position and orientation, I think you should be looking at the `tool_pose_x/y/z` and the `tool_pose_theta_x` from the [BaseFeedback](https://github.com/Kinovarobotics/kortex/blob/master/api_cpp/doc/markdown/messages/BaseCyclic/BaseFeedback.md). Orientation should be the same, but position might be slightly off because the center point of the interconnect is not the same as the camera position. 

Maybe I do not understand correctly what you want and what you are trying to do. If that's the case, can you please elaborate to help me understand?

Best,
Felix

jabbershort, 2023-01-04:
Hi @felixmaisonneuve,

Thanks for the response. I think I have mixed up the parameters here, I was trying to get the position of the camera color frame w.r.t. the base, which as you say is probably easier to do from offsetting the tool_pose from base feedback. which I'll do. Looking at the output from the camera ExtrinsicParameters, I think you are correct, I think its the relationship between the two camera frames.

closed 2023-01-04 by jabbershort