Kinovarobotics/Kinova-kortex2_Gen3_G3L #143: GEN3 simulation in GAZEBO

WHuangEI, 2023-01-15:
## Summary

I want to use GEN3 in GAZEBO to carry out some simulation research, and encountered some problems. 
1)The gripper can 't hold objects, I learned that plugins were needed, and I found them in the third-party file, but I didn 't know how to use them. 
2）and where can I find some codes or demo scripts of GEN3 simulation  in GAZEBO ? Recently, due to the inability to work with real robots, the first attempt to simulate GEN3 in GAZEBO puzzled me how to start.

## Use case

Why is this feature needed?

What are the use case covered by this feature?

## Alternative

Do you have an idea how on how best to implement this enhancement?

If this feature is not implemented, do you have a workaround?

## Possible issues

Anything that could be affected by the new feature.

## Additional context

Any other information you would like us to know about the requested enhancement.

felixmaisonneuve, 2023-01-18:
HI @WHuangEI,

Like you said, you need the third-party folder to grasp objects in gazebo. This is compiled automatically and it should work out of the box. You could share your logs from the `roslaunch spawn_kortex_robot.launch ...` command to see if everything was started correctly. 
When you close your gripper on and object, you wait 2-3 seconds and a log should appear that says the object was grasped by the gripper. 

There is no demo specifically for gazebo. You can control the arm in gazebo the same way you control a real arm. Not all functions are available however. All the available functions are listede here : https://github.com/Kinovarobotics/ros_kortex/tree/noetic-devel/kortex_examples#the-ways-to-use-a-kortex-arm-with-ros
This means the same program that was working on a real arm should also work in gazebo as long as your are using services that are implemented for the simulation

Best,
Felix

WHuangEI, 2023-02-15:
thank for your kind answer, it helps a lot, i will close the issue.

closed 2023-02-15 by WHuangEI