Kinovarobotics/Kinova-kortex2_Gen3_G3L #145: Inaccurate end-effector wrench measurements

ShaoxiongYao, 2023-01-16:
## Summary

Improve end-effector force/torque measurement accuracy.

## Use case

Detect end-effector collision with other objects or control contact force.
Currently the end-effector APIs `base_feedback.base.tool_external_wrench_force_x/y/z` and `base_feedback.base.tool_external_wrench_torque_x/y/z` are not sufficiently accurate.

## Alternative

Maybe there is a way to calibrate the wrench measurement?

## Possible issues

Should not have negative impacts on other features.

## Additional context

Here is an experiment I have on the current end-effector wrench measurements API, where the robot arm periodically moves forward and backward in the x-axis. The first figure show the relation between x-axis coordinate over time:
[image]

The second figure shows the force in the x direction measured at each x coordinate, in a scatter plot:
[image]

We may see depending on different moving directions, the force measurements are quite different. For more information, there is the robotiq85 gripper mounted on the end-effector and there is no object grasped or hanging on the gripper. So I expected zero force and torque measurements of the end-effector wrench.

[image]

ShaoxiongYao, 2023-01-16:
I attach the code block to read the measurements here:

```
with utilities.DeviceConnection.createTcpConnection(args) as router:

        with utilities.DeviceConnection.createUdpConnection(args) as router_real_time:

            # Create required services
            base = BaseClient(router)
            base_cyclic = BaseCyclicClient(router_real_time)

            input("INFO: press enter to start collecting data")

            for _ in range(tot_num_steps):
                iter_start_time = time.time()
                base_feedback = BaseCyclic_pb2.Feedback()
                base_feedback = SendCallWithRetry(base_cyclic.RefreshFeedback, 3)

                ee_x = base_feedback.base.tool_pose_x             # (meters)
                ee_y = base_feedback.base.tool_pose_y             # (meters)
                ee_z = base_feedback.base.tool_pose_z             # (meters)
                ee_theta_x = base_feedback.base.tool_pose_theta_x # (degrees)
                ee_theta_y = base_feedback.base.tool_pose_theta_y # (degrees)
                ee_theta_z = base_feedback.base.tool_pose_theta_z # (degrees)
                ee_pose_vec = np.array([[ee_x, ee_y, ee_z, ee_theta_x, ee_theta_y, ee_theta_z]])
                ee_pose_ary = np.append(ee_pose_ary, ee_pose_vec, axis=0)

                f_x = base_feedback.base.tool_external_wrench_force_x
                f_y = base_feedback.base.tool_external_wrench_force_y
                f_z = base_feedback.base.tool_external_wrench_force_z

                tau_x = base_feedback.base.tool_external_wrench_torque_x
                tau_y = base_feedback.base.tool_external_wrench_torque_y
                tau_z = base_feedback.base.tool_external_wrench_torque_z

                f_vec = np.array([[f_x, f_y, f_z, tau_x, tau_y, tau_z]])
                wrench_ary = np.append(wrench_ary, f_vec, axis=0)
```

felixmaisonneuve, 2023-01-18:
Hi @ShaoxiongYao,

external tool wrench measurement is not super accurate because it is computed from all joint torques. I will link a comment I did that is a bit more detailed : https://github.com/Kinovarobotics/kortex/issues/125#issuecomment-1246742330

There was also some unofficial tests we did and we got an accuracy of ~8% in a static position : https://github.com/Kinovarobotics/kortex/issues/87#issuecomment-1096882756

Also, since your are measuring while moving the arm, you will get a wrench that is very innacurate (a lot worse than 8%). 

Since wrench is computed from the torque value of each joint, the only thing you could do is reset the torque offset of each joint ("setZeroTorque"). However, this will probably make little to no difference, especially if your are moving the arm. 

Unfortunately, the arm was not tuned for this specific type of use, so there is not much you can do about it. 

Sorry for the inconveniance, 
Felix

ShaoxiongYao, 2023-01-18:
Thank you for your reply! I see this is not a good use case given the current hardware setup. May I ask if there are recommended end-effector FT sensors that can be directly mounted on the Kinova arm? Like I see a question on your official webpage but I did not see an answer: https://www.kinovarobotics.com/resource/can-i-use-the-robotiq-ft-300-sensor-with-gen3. I am looking forward to your help :)

felixmaisonneuve, 2023-01-19:
Can you explain to me how you got to that page? When looking at the FAQ, I can access the answer. I am asking because I couldn't find it. We could probably address this in our webpage so the answer is displayed.

Anyway, here is the answer
[image]

Can I use the Robotiq FT-300 sensor with Gen3?
>Mounting the FT-300 on the Gen3 requires a mechanical flange sold separately by Robotiq that matches the bolt pattern on the interconnect. However, the FT-300 covers all connectors on the robot's interconnect module, so all communication and wiring, including both the cables of the FT-300 and gripper, will need to be handled outside of the robot.

ShaoxiongYao, 2023-01-21:
Thank you for your message!

[image]

I am not sure why I cannot see the answer on my side. FYI, I am visiting this webpage on Windows using Chrome, and I also disabled the AD blocker. I am good to close the issue now. Thanks!

felixmaisonneuve, 2023-01-24:
Alright, thank you very much. 

I will share this information and hopefully we will be able to fix this on the website

closed 2023-01-24 by felixmaisonneuve