Kinovarobotics/Kinova-kortex2_Gen3_G3L #147: Arm jerks when switching from high-level to low-level servoing

madan96, 2023-02-06:
## Description

I am currently looking at a task where I am tracking two sequential trajectories. For the first one, I'm using velocity control with high-level servoing. Once that's done, I switch to low-level servoing and perform effort control to track the second trajectory. During the transition, I pass gravity compensation torques to ensure that the robot is stable before the robot starts following the second trajectory. Unfortunately,  even with the gravcomp torques, the robot experiences a slight jerk when switching from high-level to low-level. I would like to know if there are any suggestions on things I can do to get rid of this jerk or minimize it?

## Version
Gen3 7 DoF with KortexAPI : 2.5.0

felixmaisonneuve, 2023-02-22:
Hi @madan96,

Do you have an update on this?
Did you manage to resolve your issue?

Best,
Felix

martinleroux, 2024-09-05:
I will be closing this issue due to inactivity.

closed 2024-09-05 by martinleroux