Kinovarobotics/Kinova-kortex2_Gen3_G3L #149: GEN3's URDF file

WHuangEI, 2023-02-15:
## Summary
I am using the URDF file to install the GEN3 on a mobile base and tried to view the model in RVIZ, but the error was as follows :
hwww@hwww-OMEN-by-HP-Laptop-15-dc1xxx:~/gen3_ws$ rviz rviz
[ INFO] [1676453112.861655301]: rviz version 1.13.29
[ INFO] [1676453112.861691100]: compiled against Qt version 5.9.5
[ INFO] [1676453112.861699124]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1676453112.885707197]: Forcing OpenGl version 0.
[ INFO] [1676453113.022364107]: Stereo is NOT SUPPORTED
[ INFO] [1676453113.022411276]: OpenGL device: Mesa DRI Intel(R) UHD Graphics 630 (CFL GT2)
[ INFO] [1676453113.022437905]: OpenGl version: 3.0 (GLSL 1.3).
[ERROR] [1676453120.867190893]: Could not load resource [package://kortex_description/arms/gen3/meshes/Bracelet_Link.STL]: Unable to open file "package://kortex_description/arms/gen3/meshes/Bracelet_Link.STL".
[ERROR] [1676453120.867827151]: Could not load resource [package://kortex_description/arms/gen3/meshes/Bracelet_Link.STL]: Unable to open file "package://kortex_description/arms/gen3/meshes/Bracelet_Link.STL".
[ERROR] [1676453120.869205918]: Could not load resource [package://kortex_description/arms/gen3/meshes/ForeArm_Link.STL]: Unable to open file "package://kortex_description/arms/gen3/meshes/ForeArm_Link.STL".
[ERROR] [1676453120.869747542]: Could not load resource [package://kortex_description/arms/gen3/meshes/ForeArm_Link.STL]: Unable to open file "package://kortex_description/arms/gen3/meshes/ForeArm_Link.STL".
[ERROR] [1676453120.870574529]: Could not load resource [package://kortex_description/arms/gen3/meshes/HalfArm1_Link.STL]: Unable to open file "package://kortex_description/arms/gen3/meshes/HalfArm1_Link.STL".
[ERROR] [1676453120.871066437]: Could not load resource [package://kortex_description/arms/gen3/meshes/HalfArm1_Link.STL]: Unable to open file "package://kortex_description/arms/gen3/meshes/HalfArm1_Link.STL".
[ERROR] [1676453120.871786090]: Could not load resource [package://kortex_description/arms/gen3/meshes/HalfArm2_Link.STL]: Unable to open file "package://kortex_description/arms/gen3/meshes/HalfArm2_Link.STL".
[ERROR] [1676453120.872293446]: Could not load resource [package://kortex_description/arms/gen3/meshes/HalfArm2_Link.STL]: Unable to open file "package://kortex_description/arms/gen3/meshes/HalfArm2_Link.STL".
[ERROR] [1676453120.872941722]: Could not load resource [package://kortex_description/arms/gen3/meshes/Shoulder_Link.STL]: Unable to open file "package://kortex_description/arms/gen3/meshes/Shoulder_Link.STL".
[ERROR] [1676453120.873487275]: Could not load resource [package://kortex_description/arms/gen3/meshes/Shoulder_Link.STL]: Unable to open file "package://kortex_description/arms/gen3/meshes/Shoulder_Link.STL".
[ERROR] [1676453120.874127240]: Could not load resource [package://kortex_description/arms/gen3/meshes/SphericalWrist1_Link.STL]: Unable to open file "package://kortex_description/arms/gen3/meshes/SphericalWrist1_Link.STL".
[ERROR] [1676453120.874756705]: Could not load resource [package://kortex_description/arms/gen3/meshes/SphericalWrist1_Link.STL]: Unable to open file "package://kortex_description/arms/gen3/meshes/SphericalWrist1_Link.STL".
[ERROR] [1676453120.875549398]: Could not load resource [package://kortex_description/arms/gen3/meshes/SphericalWrist2_Link.STL]: Unable to open file "package://kortex_description/arms/gen3/meshes/SphericalWrist2_Link.STL".
[ERROR] [1676453120.876024410]: Could not load resource [package://kortex_description/arms/gen3/meshes/SphericalWrist2_Link.STL]: Unable to open file "package://kortex_description/arms/gen3/meshes/SphericalWrist2_Link.STL".
[ERROR] [1676453120.972629977]: Could not load resource [package://kortex_description/arms/gen3/meshes/base_link.STL]: Unable to open file "package://kortex_description/arms/gen3/meshes/base_link.STL".
[ERROR] [1676453120.972732924]: Could not load resource [package://kortex_description/arms/gen3/meshes/base_link.STL]: Unable to open file "package://kortex_description/arms/gen3/meshes/base_link.STL".

I have sourced the ws, and i don 't know where the problem is. Are the stl files damaged? How can I solve the problem ？

## Use case

Why is this feature needed?

What are the use case covered by this feature?

## Alternative

Do you have an idea how on how best to implement this enhancement?

If this feature is not implemented, do you have a workaround?

## Possible issues

Anything that could be affected by the new feature.

## Additional context

Any other information you would like us to know about the requested enhancement.

felixmaisonneuve, 2023-02-17:
Hi @WHuangEI,

If I compare the path from one of the error message `kortex_description/arms/gen3/meshes/Bracelet_Link.STL`, I see it is missing `6dof` or `7dof` after `gen3`. It should look something like `kortex_description/arms/gen3/7dof/meshes/Bracelet_Link.STL`

Can you confirm if your directory is correct?

Best,
Felix

WHuangEI, 2023-02-18:
Yes, I've tried adding 7dof but still got this error, and directory is correct. And 'Could not load resource' and 'Unable to open file' show that the computer can find the file but cannot load the contents of the file，so I don 't think there 's a problem with the path and don't know where the problem is.

felixmaisonneuve, 2023-02-22:
It says 'Could not load resource' and 'Unable to open file' most likely because it cannot find the files in your package. Maybe your package was not built correctly. 

What changes did you make? What URDF file are you using? How is it integrated with your base's URDF and ROS module?

martinleroux, 2024-07-17:
I will be closing this issue due to lack of activity. If you need further assistance, please raise an issue in the ros_kortex or ros2_kortex repositories (whichever is applicable), or reach out to us at support@kinova.ca.

closed 2024-07-17 by martinleroux