Kinovarobotics/Kinova-kortex2_Gen3_G3L #151: C++ can we have kortex as separate library that can be installed?

rojikada, 2023-02-22:
## Summary
Hello, for my purpose I would like to use just the library, without the examples. However with the current setup it is not easy to separate as everything happens in the examples folder.
Could you make this library installable, so I can build and install it - and in separate CMakeLists.txt use it just like the other packages such as `find_package(Eigen3 3.3 REQUIRED NO_MODULE)` or `find_package(qualisys_cpp_sdk REQUIRED)`?

With the current setup it seems quite time consuming and unintuitive, that the installation happens inside examples folder.

## Use case

Why is this feature needed?
Improve build times, remove clutter from the kortex build, easy to use.

What are the use case covered by this feature?
Separate installation and examples. Easy to use library.

## Alternative

Do you have an idea how on how best to implement this enhancement?
No.

If this feature is not implemented, do you have a workaround?
Yes, I currently copied and modified the CMakeLists.txt. However, it has to compile every time I want to use this library, which is not very convenient.

## Possible issues

Anything that could be affected by the new feature.

## Additional context

Any other information you would like us to know about the requested enhancement.

felixmaisonneuve, 2023-02-22:
Hi @rojikada,

I am not sure what you mean when you say you need to use the library, but not the examples. What are you trying to do?
The KortexAPI is already a package that you need to download (either with conan or manually) to build the examples.
with conan : 
https://github.com/Kinovarobotics/kortex/blob/392c312d57912e9645c70cb3e783bdcdb89371bd/api_cpp/examples/CMakeLists.txt#L118-L129

manually (you download first) :
https://github.com/Kinovarobotics/kortex/blob/392c312d57912e9645c70cb3e783bdcdb89371bd/api_cpp/examples/CMakeLists.txt#L180-L185
then link it :  
https://github.com/Kinovarobotics/kortex/blob/392c312d57912e9645c70cb3e783bdcdb89371bd/api_cpp/examples/CMakeLists.txt#L201-L205 

We are using a single CMakeLists.txt file inside our example folder because it makes it easier to add new examples (you only need to add a new folder) and this reduces the amount of copied CMakeLists.txt files (only one file instead of one per folder). It also makes it easier to use: you build everything the first time (takes a bit of time), but then you have all the examples and you can use them all as you wish. 

I think it makes sense to have only one CMakeLists.txt file since these are examples, so you shouldn't have to build them often. If you want to develop a program using the Kortex API, you should implement your own file structure separately from this one. 

Does that help you? 
I am not sure I fully understand what you are trying to do

Best,
Felix

rojikada, 2023-02-23:
I would like to use kortex as a library such as Eigen (used as Eigen::Vector3d by including <Eigen/Dense>). This library can be either installed directly as a package or build as a separate package and then I just use find_package(...)) to link it. But it is not re-building it every time.

In my code I use  basic function such as setting the robots end-effector velocity and setting velocity of the gripper. For that I use:
```
#include <BaseClientRpc.h>
#include <SessionManager.h>
#include <RouterClient.h>
#include <TransportClientTcp.h>
```

But those don't seem to be part of the library, so I have to compile with the following includes:
  ```
# Add Include Directories
  include_directories(${KORTEX_DIR}include)
  include_directories(${KORTEX_DIR}include/client)
  include_directories(${KORTEX_DIR}include/common)
  include_directories(${KORTEX_DIR}include/messages)
  include_directories(${KORTEX_DIR}include/client_stubs)
```

And I would like to avoid recompiling those includes every time as it takes about half time of the total compilation time + throws quite a few warnings about is_pod usage from old Protobuf library.

martinleroux, 2024-07-17:
Hi @rojikada ,

I am revisiting all of our opened issues in the repo and I notice that this one was left unanswered since Felix left.

In short, as Felix said previously, Kortex API is a single library - you can see this by noting the package contains a single .lib/.dll file. It's just the include directories that are split. The include subfolders in ${KORTEX_DIR} do not include any of the examples in the repo. If you are building your own application without the examples, you will still need to include all of these to have access to the full scope of the API.

closed 2024-09-05 by martinleroux