Kinovarobotics/Kinova-kortex2_Gen3_G3L #155: urdf error

hejian1w, 2023-03-03:
## Description
Data in ROS_KORTEX_NOETIC_DEVEL  kortex_description->arms->gen3 -> GEN3_YRDF_V12.urdf;  The urdf data in 7dof is inconsistent with the document （Homogeneous transform matrices - 7 DoF spherical wrist）, I want to use this urdf data to make IK and FK, I should refer to the data of the real robot arm GEN3, is this a bug

  <joint name="Actuator1" type="continuous">
    <origin xyz="0 0 0.15643" rpy="3.1416 2.7629E-18 -4.9305E-36" />

The XYZ data has some errors

hejian1w, 2023-03-03:
> 
For example
urdf：  
<joint name="Actuator1" type="continuous">
    <origin xyz="0 0 0.15643" rpy="3.1416 2.7629E-18 -4.9305E-36" />

doc：
  0.1564

rojikada, 2023-03-06:
If you want FK/IK you can use DH parameters. The base transformation is:
   ```
    1, 0,  0,   0,
    0,  -1, 0,   0,
    0,  0,  -1,  0,
    0,  0,  0,   1
```

and the end-effector is without any rotation and Z-axis is +0.120m.

hejian1w, 2023-03-07:
I mean, in the document, the offset data is 4 decimal places, like 0.1564, while on the UDF file, it is 5 decimal places, 0.1564

hejian1w, 2023-03-07:
I mean, in the document, the offset data is 4 decimal places, like 0.1564, while on the urdf file, it is 5 decimal places, 0.15643. What set of data should I end up with?

hejian1w, 2023-03-07:
> 
When I use the example in the document, the calculated fk is inconsistent with the real robotic arm end data, which should be caused by some details

hejian1w, 2023-03-11:
> If you want FK/IK you can use DH parameters from the documentation PDF. The base transformation is:
> 
> ```
>  1, 0,  0,   0,
>  0,  -1, 0,   0,
>  0,  0,  -1,  0,
>  0,  0,  0,   1
> ```
> 
> and the end-effector is without any rotation and Z-axis is +0.120m.

What is the source of the data?

rojikada, 2023-03-11:
It comes from DH parameters inside the documentation. This is the 0-th parameter converted into transformation matrix.

hejian1w, 2023-03-13:
> It comes from DH parameters inside the documentation. This is the 0-th parameter converted into transformation matrix.

Could you please supply a screenshot from Docs? I only found one matrix which contains transformation matrix of the base matrix you provided

rojikada, 2023-03-13:
[image]

hejian1w, 2023-03-14:
> [image]
> 
> from EN-UG-014-Gen3-Ultralight-user-guide-r8.pdf

My matrix is based on this, but your 0.120 does not have a value that represents the source of this data. Can you give me a contact information

rojikada, 2023-03-14:
I see, 0.12 comes from different place:
[image]

hejian1w, 2023-03-14:
Thanks！I will check the data again.

felixmaisonneuve, 2023-04-03:
Hi @hejian1w,

Were you able to find all the information you wanted in the Gen3 user guide?

Best,
Felix

martinleroux, 2024-09-05:
Hi all,

I will be closing this issue since it seems to have been solved above.

However, for future reference to anyone ending up here, you should know that the URDF is the more accurate data - we limit the number of decimal places used in the User Guide for repeatability, but the URDF is generated directly from the CAD.

However, should you need so much precision that this very small difference matters, then you are going to face another issue in that the URDF model is an ideal one and does not take into consideration assembly tolerances. Each robot is calibrated, which explains why manually computing the kinematics outputs a value that differs from what is reported by the robot in the Web App. If you need information regarding the calibration parameters of your specific unit, please contact support@kinova.ca with the serial number of your arm (found on the sticker at the base - the value in the Web app is the SN of the controller, which is not what we are looking for) and we will find the calibration file of your robot in our archives.

Cheers

closed 2024-09-05 by martinleroux