Kinovarobotics/Kinova-kortex2_Gen3_G3L #157: Needs of rotor inertia value for efficient torque control on Gen3 robots

mathieu-celerier, 2023-03-07:
I am currently working on Kinova Gen3 using torque control.
To improve the torque control I would like to include the dynamical effect of rotor inertia in my framework.

Is it possible to know what motors are used in the Gen3 robot?

If not, could you please provide the following:
- Rotor inertia
- Gear Ratio

For both;
- Large actuators
- Small actuators

Best regards,

Mathieu Celerier

felixmaisonneuve, 2023-03-17:
Hi @mathieu-celerier,

We cannot share the actuator model, but you can contact the support team at support@kinova.ca and they might be able to share the inertia and gear ratio

Regards,
Felix

closed 2023-03-17 by felixmaisonneuve

juandavid212, 2023-04-18:
Hi @mathieu-celerier,
Did you manage to implement this efficient torque control? I'm planning to implement something similar. Would you mind sharing your code?