Kinovarobotics/Kinova-kortex2_Gen3_G3L #162: problem when running 110-waypoints example code

WHuangEI, 2023-04-13:
## Description
when running 110-waypoints example code, some errors occurred，how can I solve this ?

~/gen3_ws/src/kortex/api_python/examples/110-Waypoints$ python3 02-send_cartesian_waypoint_trajectory.py 
Logging as admin on device 192.168.1.10
Moving the arm to a safe position
EVENT : ACTION_START
EVENT : ACTION_END
Safe position reached
Traceback (most recent call last):
  File "02-send_cartesian_waypoint_trajectory.py", line 209, in <module>
    exit(main())
  File "02-send_cartesian_waypoint_trajectory.py", line 204, in main
    success &= example_trajectory(base, base_cyclic)
  File "02-send_cartesian_waypoint_trajectory.py", line 140, in example_trajectory
    result = base.ValidateWaypointList(waypoints)
  File "/home/hwww/.local/lib/python3.6/site-packages/kortex_api/autogen/client_stubs/BaseClientRpc.py", line 2156, in ValidateWaypointList
    result = future.result(options.getTimeoutInSecond())
  File "/usr/lib/python3.6/concurrent/futures/_base.py", line 432, in result
    return self.__get_result()
  File "/usr/lib/python3.6/concurrent/futures/_base.py", line 384, in __get_result
    raise self._exception
kortex_api.Exceptions.KServerException.KServerException: Server error name=ERROR_PROTOCOL_SERVER, sub name=UNSUPPORTED_METHOD => Error details were not received from server. 

## Version

At a minimum, provide the Kortex API and Kortex-enabled device versions.

KortexAPI :

Kortex Device :

## Steps to reproduce

1. *step 1...*
2. *step 2...*

## Code example

```
    # paste code here #
```

## Expected behavior

Describe the expected behavior (in regard to the bug described above).

## Any other information

Any other information you believe the developers need to know.

felixmaisonneuve, 2023-04-13:
Hi @WHuangEI,

It looks like the `ValidateWaypointList` is not recognize by the arm. Can you make sure you are using a SWU version >=2.3?
`ValidateWaypointList` was added in the Kortex Release 2.3, so if your arm is still using an older firmware package (<=2.2), it will fail the `UNSUPPORTED_METHOD` error message

Regards,
Felix

WHuangEI, 2023-04-14:
Thank you for your reply and answer.  I downloaded the whole file and configured the robot around December 2022. The version is supposed to be > = 2.3.0. It confuses me and how can I check the API version I 'm using now and update it without damage to the other documents if my version < 2.3.0

WHuangEI, 2023-04-14:
And  I thought After installing the whl file, I was able to run some example codes as instructed by readme.md, but there was nothing about installing SWU, and it seems that I did not download the install SWU file anymore.

WHuangEI, 2023-04-14:
I entered the Kinova Web Application, and found that the Web App Version is 2.8.0-r.33, Web Client API Version is 2.2.0-r.27, Firmware Bundle is 2.2.0-r.51, and Armbase 2.2.0-r.58, all Actuators 2.2.0-15, and Vision is 2.2.0-r.22. Can i just press the 'upload' to upgrade it? is that a solution?

WHuangEI, 2023-04-14:
I entered the Kinova Web Application, and found that the Web App Version is 2.8.0-r.33, Web Client API Version is 2.2.0-r.27, Firmware Bundle is 2.2.0-r.51, and Armbase 2.2.0-r.58, all Actuators 2.2.0-15, and Vision is 2.2.0-r.22. Can i just press the 'upload' the 'Gen3-2.4.0.swu' file just downloaded today to upgrade my arm? is that a solution?

WHuangEI, 2023-04-14:
I tried that and new errors occurred like this:
hwww@hwww-OMEN-by-HP-Laptop-15-dc1xxx:~/gen3_ws/src/kortex/api_python/examples/110-Waypoints$ python3 02-send_cartesian_waypoint_trajectory.py 
Logging as admin on device 192.168.1.10
Moving the arm to a safe position
Traceback (most recent call last):
  File "02-send_cartesian_waypoint_trajectory.py", line 210, in <module>
    exit(main())
  File "02-send_cartesian_waypoint_trajectory.py", line 204, in main
    success &= example_move_to_home_position(base)
  File "02-send_cartesian_waypoint_trajectory.py", line 55, in example_move_to_home_position
    action_list = base.ReadAllActions(action_type)
  File "/home/hwww/.local/lib/python3.6/site-packages/kortex_api/autogen/client_stubs/BaseClientRpc.py", line 641, in ReadAllActions
    result = future.result(options.getTimeoutInSecond())
  File "/usr/lib/python3.6/concurrent/futures/_base.py", line 432, in result
    return self.__get_result()
  File "/usr/lib/python3.6/concurrent/futures/_base.py", line 384, in __get_result
    raise self._exception
kortex_api.Exceptions.KServerException.KServerException: Server error name=ERROR_DEVICE, sub name=DATABASE_ERROR => no type information

felixmaisonneuve, 2023-04-17:
So you did the right thing.

Now, for the database error, this is probably because you have an older arm and the format of an "Action" changed. A Factory Reset will fix the issue, but you will lose all your protection zones and your saved Actions

felixmaisonneuve, 2023-04-17:
Sorry, I clicked the wrong button

I can refer you to #100, which was the same issue

Just be carefull when doing a actory reset. It will wipe all your configurations, your protection zones, your actions ...

WHuangEI, 2023-04-27:
Thank you so much，I did it according to your guidance, and the problem was solved ! 
One more thing, I want to implement admittance control in my own code, although there are already admittance control mode that can be used directly. I calculate the acceleration, speed and position according to the admittance control algorithm. Which API or function can I use to make the manipulator perform them better ? I tried to use PlayCartesianTrajectoryPosition() and PlayJointTractory(), but there was an obvious stumble that was not smooth. How does the admittance control mode provided by you can make the robot so smooth？

felixmaisonneuve, 2023-05-10:
I would suggest to use the [Low Level](https://github.com/Kinovarobotics/kortex/blob/master/linked_md/cpp_servoing_modes.md#low-level-servoing-mode) servoing mode instead of high level functions. 

This will allow you to control each actuator directly without interracting the with the base. You will have faster reponse time (low level works at 1000Hz compared to 40Hz) and will be able to send velocity/torque/position and/or current commands to each actuator individually. You will lose some high level data that the base computes, but I do not think you will need this data anyway. Low level also requires you to send a command each ms to the arm, so C++ is (almost) mandatory

Here are two examples that might be usefull to get you going : 
https://github.com/Kinovarobotics/kortex/blob/master/api_cpp/examples/200-Actuator_low_level_control/01-actuator_low_level_velocity_control.cpp
https://github.com/Kinovarobotics/kortex/blob/master/api_cpp/examples/108-Gen3_torque_control/01-torque_control_cyclic.cpp

Hope this helps

WHuangEI, 2023-05-11:
Thank you very much, I will try your advice later and I will close this issue then.

martinleroux, 2024-09-05:
I will be closing this issue due to the original question being answered and the rest of the thread being inactive.
If you have further problems with the second topic, please open a new issue so the title matches what is being discussed.

closed 2024-09-05 by martinleroux