Kinovarobotics/Kinova-kortex2_Gen3_G3L #167: Camera Calibration

RobaczeQ, 2023-04-27:
## Summary

Hi again. 
I was trying to align point clouds from different points to have ability to estimate pose/orientation of some cubes using provided intrinsic and extrinsic camera parameters, but they are slightly misaligned. I was trying to use Dynamic Calibration Tool provided by Intel® RealSense™, but it doesn't support calibrating it through external systems connected by Ethernet. 

Can I connect to camera by USB somehow? 
Can I access left, right and IR sensor separately or only 2d map of depth?

> 4.3 Connect camera to host system that whilst the camera that is to be calibrated will usually be connected to the host computer via a USB cable, "on systems with camera integrated it's already connected, so no extra step is required".

Can I connect to Kinova internal system and install calibration there? 
Is there a plan to add camera calibration to internal software like other in `enum CalibrationItem`?
Any other ideas on how to get around this problem? (Other than suggested in #116)

## Use case

Creating precise point clouds of environment.

felixmaisonneuve, 2023-05-10:
Hi @RobaczeQ,

You cannot connect your camera by USB.
You cannot connect to different sensor seperately. You only ave access to either Color or Depth sensor. 

I do not think you can install a calibration file on the arm.
There is nothing currently planned to improve/fix the vision module.

I hae very limited knowledge on the vision module, and I know many people are having issues trying to align depth and color streams. I do not know if there is anything you could do.

There is this one repo in the #116 issue that is also available in the feature/calibration-program branch of our ros_kortex_vision repo, but it requires you to use ROS and it doesn't actually align the streams inside the arm, I think it only aligns it on the RViZ feed.

So yeah, very little I can do to help you,
Best,
Felix

RobaczeQ, 2023-05-10:
Thanks for information.

closed 2023-05-10 by RobaczeQ

felixmaisonneuve, 2023-05-25:
After verification, it is possible to disaemble the vision module and connect to the Intel depth sensor using USB-C