Kinovarobotics/Kinova-kortex2_Gen3_G3L #168: Inaccurate robot's movement in Cartesian mode

pouyan-asg, 2023-05-08:
## Description
Hi, 

The problem I encountered is that when I want to control the robot in Cartesian mode, robot can not follow all waypoints accurately. In other words, consider robot receive two points in XYZ plane in which X and Z are constant. The difference between Y points are 14 inch but after touching the first point, robot touches the second point in 12.5 inches (1.5 inches less than expected). It seems a scale down happened but I checked all points and robot's input is true. 
In this example, if robot wants to touch exactly 14 inches differences in real-time, I should send the second point with location more that 14 inches. 
In order to be clearer, below is output of monitoring for two points with 14 inches differences. 

```
[
    {
        "title": "base",
        "flat": true,
        "data": [
            {
                "title": "Operating Mode",
                "value": "Running",
                "safety": {
                    "status": 3,
                    "safeties": []
                }
            },
            {
                "title": "Control Mode",
                "value": "monitoring.controlMode.CARTESIAN_WAYPOINT_TRAJECTORY",
                "safety": {
                    "status": 3,
                    "safeties": []
                }
            },
            {
                "title": "Servoing Mode",
                "value": "Single Level",
                "safety": {
                    "status": 3,
                    "safeties": []
                }
            }
        ]
    },
    {
        "title": "actuators",
        "grid": true,
        "data": [
            {
                "isHeaderColumn": true,
                "title": "actuator #1"
            },
            {
                "isHeaderColumn": true,
                "title": "actuator #2"
            },
            {
                "isHeaderColumn": true,
                "title": "actuator #3"
            },
            {
                "isHeaderColumn": true,
                "title": "actuator #4"
            },
            {
                "isHeaderColumn": true,
                "title": "actuator #5"
            },
            {
                "isHeaderColumn": true,
                "title": "actuator #6"
            },
            {
                "isHeaderColumn": true,
                "title": "actuator #7"
[... 506 lines cut ...]
                "value": "0 °",
                "safety": {
                    "status": 3,
                    "safeties": []
                }
            },
            {
                "title": "Commanded Tool Pose Theta Z",
                "value": "90 °",
                "safety": {
                    "status": 3,
                    "safeties": []
                }
            }

```

As you can see, the points robot receive are correct but robot movements is not. 
Is there a setting issue in software side or it relates to hardware joints inaccuracy?

Thank you very much in advance for your time

## Expected behavior

accurate movements toward waypoints

felixmaisonneuve, 2023-05-10:
Hi @pouyan-asg,

I am not entirely sure I understand your issue. 
According to the feedback you shared, the arm moved to the corretc position for both points and the distance between them is about 14inches.
Based only on the numbers, everything seems fine.

Or maybe this is the issue, numbers are fine, but the arm physically moves only 12.5 Inches?

Can you elaborate a bit more please.

Best,
Felix

pouyan-asg, 2023-05-11:
Hi @felixmaisonneuve ,

That's true. based on numbers, it shows fine but arm physically moves only 12.5 Inches (difference between actual number and desired value)

pouyan-asg, 2023-05-16:
Kinova company directly solved the issue. Thanks

closed 2023-05-16 by pouyan-asg

felixmaisonneuve, 2023-05-18:
Out of curiosity, what was the issue?

pouyan-asg, 2023-05-22:
That was a problem in one joint. After various tests, it showed that was a malfunction.