Kinovarobotics/Kinova-kortex2_Gen3_G3L #171: Saving Gen3 logs

pouyan-asg, 2023-05-26:
## Summary

I want to save robots logs and errors in a separate file. I added simple saving results as a txt file in "kortex_api.Exceptions" codes. I am wondering is there any short code or feature for saving all logs, performance data and so on or the way I am doing is correct? Also, I should mention that I am recording Kinematic_limits_list and Waypoint_list through this simple way (saving as txt file).

## Use case

Why is this feature needed? for easier tracking, troubleshooting and asking questions from Kinova company

## Alternative

Do you have an idea how on how best to implement this enhancement? I think there should be a short code or sth like that to record and save all essential data like Snapshot Data in Webpage

If this feature is not implemented, do you have a workaround? I would like to complete it.

## Possible issues

I don't think any issue will be arisen because it is just saving all logs.

felixmaisonneuve, 2023-05-30:
Hi @pouyan-asg,

Can you elaborate on "logs" and "errors"? How do you define them? What is the difference between them to you?

Our API can communicate at a rate of 1000Hz, this means you can monitor the state of every component every milliseconds. This is some kind of "logs" that you can save if that is something you need. This is not the kind of use case that is always needed however and it is up to users to implement what they need.

Same goes for errors. If an error occurs, an action will be required and user can log what happened if needed. 

Of course, all of this assumes you are using the API directly and not the WepApp, but I assume this will already be the case if you need logging. There is no mechanism in the WebApp to monitror the arm and log the information. 

Maybe I do not understand correctly what you say, if that's the case, can you elaborate on what you want too?

Best
Felix

pouyan-asg, 2023-06-01:
Hi Felix,

About saving all necessary data or errors, I wrote manually some simple lines to save robot's parameters that I need as text file. However, I am not sure is there any pre-defined function for recording robot's logs (normal operations or during errors). 
For example, every time robot is working, all necessary data like speed, velocity, positions, errors (if happened), etc record and save a JSON or Text file. 

I am not sure, I could explain clear or not. 

thanks for your time

felixmaisonneuve, 2023-06-01:
I think I understand what you want.

No, we do not provide a function or something to log. We provide [Cyclic data](https://github.com/Kinovarobotics/kortex/blob/master/api_cpp/doc/markdown/messages/BaseCyclic/Feedback.md#) and [Notifications](https://github.com/Kinovarobotics/kortex/blob/master/api_cpp/doc/markdown/summary_pages/Base.md) (see every `OnNotifcation...` call) where users can select and log themselves only the information they need. 

Since every use case is different, there is no point to provide a logging feature. Instead, we provide ways to fetch any information that one might need and users can implement their own specific logging feature. That way, they can only log the data they required and log it to the specific format they need. 

Best,
Felix

closed 2023-06-01 by pouyan-asg