Kinovarobotics/Kinova-kortex2_Gen3_G3L #172: Cannot install KORTEX API

TomSAVU, 2023-05-30:
## Description

I'm receiving errors when trying to build in CLion

## Version

KortexAPI : 2.5.0

Kortex Device : Windows 10 64 bit

## Steps to reproduce

Performed all the steps in https://www.youtube.com/watch?v=JparQ69LMzY&list=PLz1XwEYRuku5rZjJWBr6SDi93jgWZ4FHL&index=3

## Any other information

The error I receive when trying to build is 
D:/KINOVA/kortex-master/api_cpp/examples/thirdParty/cxxopts/cxxopts.hpp: In member function 'void cxxopts::values::detail::SignedCheck<T, true>::operator()(bool, U, const string&)':
D:/KINOVA/kortex-master/api_cpp/examples/thirdParty/cxxopts/cxxopts.hpp:471:42: error: 'numeric_limits' is not a member of 'std'
  471 |             if (u > static_cast<U>(-std::numeric_limits<T>::min()))
      |                                          ^~~~~~~~~~~~~~
D:/KINOVA/kortex-master/api_cpp/examples/thirdParty/cxxopts/cxxopts.hpp:471:58: error: expected primary-expression before '>' token
  471 |             if (u > static_cast<U>(-std::numeric_limits<T>::min()))
      |                                                          ^
D:/KINOVA/kortex-master/api_cpp/examples/thirdParty/cxxopts/cxxopts.hpp:471:61: error: '::min' has not been declared; did you mean 'std::min'?
  471 |             if (u > static_cast<U>(-std::numeric_limits<T>::min()))
      |                                                             ^~~
      |                                                             std::min

TomSAVU, 2023-05-30:
Thanks Felix,
I'm using a version of CLion installed today, because this is what I saw in 
https://www.youtube.com/watch?v=JparQ69LMzY&list=PLz1XwEYRuku5rZjJWBr6SDi93jgWZ4FHL&index=3
and I thought it's recommended.
I will try with the instructions you recommended and using Visual Studio and I will come back.

TomSAVU, 2023-05-30:
Hi @felixmaisonneuve 

In Visual Studio 2022 it shows me 25 similar errors, first one in hash.h line 241, which is
struct hash : public GOOGLE_PROTOBUF_HASH_COMPARE <Key> {

The first error is 
C2143 syntax error: missing a ',' before '<'

Next error is
C2039 'hash compare': is not a member of 'std'

felixmaisonneuve, 2023-05-31:
For you error in CLion, it looks like this issue : https://stackoverflow.com/questions/4798936/numeric-limits-was-not-declared-in-this-scope-no-matching-function-for-call-t

Try adding `#include <memory>` in cxxopts.hpp in thirdParty folder

For your VisualStudio errors, I do not understand what you are talking about. hash.h is not a file in our repository and `GOOGLE_PROTOBUF_HASH_COMPARE` is not present in any file.

What version of cmake, make, gcc and g++ are you using?
What commit are you using?

TomSAVU, 2023-07-28:
I'm using Microsoft Visual Studio Community 2022 (64-bit) - Version 17.6.2.
Stopped following the install guide from YouTube and I used the native CMake.
Although I disabled Conan in the CMake settings, I was still receiving error messages.
I've commented all the references to Conan in conan.cmake and CMake generation finishes without errors.
When trying to build, there are errors like "Cannot open include file: 'SessionManager.h': No such file or directory" in 01-api_creation.cpp or "Cannot open include file: 'BaseClientRpc.h': No such file or directory" in 01-gripper_command.cpp

felixmaisonneuve, 2023-07-28:
Did you download the C++ API zip archive [msvc 2019](https://artifactory.kinovaapps.com/ui/repos/tree/General/generic-local-public/kortex/API/2.6.0/windows_x86-64_msvc-2019.zip) [mingw](https://artifactory.kinovaapps.com/ui/repos/tree/General/generic-local-public/kortex/API/2.6.0/windows_x86-64_mingw.zip) (make sure to take the one for your architecture). 

If you disabled conan and you are trying to build manually, the API must be located at the correct path in the kortex_api folder
https://github.com/Kinovarobotics/kortex/tree/master/api_cpp/examples#not-using-conan

This is where it will look for the library by default
https://github.com/Kinovarobotics/kortex/blob/6f557a8c65d5b5d8e83bf4966604788a05713658/api_cpp/examples/CMakeLists.txt#L171

https://github.com/Kinovarobotics/kortex/blob/6f557a8c65d5b5d8e83bf4966604788a05713658/api_cpp/examples/CMakeLists.txt#L201-L212

Krsikapa00, 2023-12-30:
Hi, @felixmaisonneuve I am also getting the same issue as @TomSAVU when I build any of the examples. The following errors coming from the hash.h file found in the Kortex_Api in "./kortex_api/include/google/stubs/hash.h":

C2143: syntax error: missing ',' before '<' Line 241
C2039 'hash_compare': is not a member of 'std'
...

And a few more. I have downloaded the msvc 2019 API archive mentioned in your comment above and followed all the steps in the C++ guide. The CMake generation completes without error and it shows it is using the right folder for the kortex_api but building any example fails.

Would you have any ideas? Let me know if you need more information.