Kinovarobotics/Kinova-kortex2_Gen3_G3L #176: using kortex_api  in ROS (low-level)

odj3460, 2023-07-10:
## Summary

I am currently utilizing two Gen3 robots for my project.

I have attempted control using the kortex_driver, but I found the control cycle to be slow, which has led me to consider using the API directly. I intend to control the robot through low-level servoing, and I also need to use additional external sensors, for which I plan to use ROS.

I am wondering if there might be any issues with controlling the robot through the API within ROS?

From what I understand based on other issues and discussions, it seems the maximum control rate with ROS is 40Hz, but I am assuming this limitation applies specifically to the kortex_driver. I am curious to know if it is possible to achieve a control rate of over 100Hz using low-level control through the API within ROS?

Additionally, I would appreciate if you could provide me with any examples or resources related to this topic.

## Use case

Dual arm manipulation with FT sensors (ROS) (1000hz)

felixmaisonneuve, 2023-07-10:
Hi @odj3460,

You can control the gen3 arm via the API within ROS, but the low-level API is not available in the current implementation of the driver. 

I suppose it would be possible to modifiy the kortex_driver to support low-level, but this is no easy task. If you manage to implement such thing, you will need to change the rate to 1000Hz. This is an API requirement and you will may experience issues if you are not communicating at exactly 1ms intervals (being above 100Hz will not be enough). 

I do not think anyone has implemented this low-level support in our ROS stack, so no resources are available on this topic. We usually recommand using the API directly when low-level is required. Low-level using the API is well documented and you can find examples on our other Kortex repo
https://github.com/Kinovarobotics/kortex/blob/master/api_cpp/examples/107-Gripper_low_level_command/01-gripper_low_level_command.cpp
https://github.com/Kinovarobotics/kortex/blob/master/api_cpp/examples/108-Gen3_torque_control/01-torque_control_cyclic.cpp
https://github.com/Kinovarobotics/kortex/blob/master/api_cpp/examples/200-Actuator_low_level_control/01-actuator_low_level_velocity_control.cpp

Best,
Felix

felixmaisonneuve, 2023-07-11:
I have been told that some people tried to increase the ROS rate in our driver and they only managed to get to ~200-300Hz, which is not enough to support low-level. This would make it impossible to implement anything in ROS to use the low-level API of the arm.

odj3460, 2023-07-12:
Then, it seems pointless to build a system to utilize the API within ROS.

As I understand, in the low-level, we are unable to use the API for the information related to the interface configuration attached to the end of the robot. 
So, does it mean that if we use the low-level API, we are unable to build a system integrated with external sensors?

felixmaisonneuve, 2023-07-12:
I am not sure what interface configuration you refer too. Can you please detail on that?

odj3460, 2023-07-14:
Ultimately, what I want to do is to integrate the Kinova api (low-level) and the FT sensor with ROS, controlling the robot at a rate of 1000hz.

The 'interface configuration' mentioned in the previous response refers to the method of connecting end-effector devices via UART or GPIO and using the related APIs.

As I previously mentioned, if the low-level API does not work within ROS in the way I want, do I need to use external sensors through the interface configuration? 
If so, do the interface-related APIs in the Kortex API operate at 1000hz?

felixmaisonneuve, 2023-07-20:
The low-level API does not work within ROS.

It is possible, even in low-level, to use the Interconnect interface. You can call the any function of the InterconnectConfig even in low-level (so at a 1000Hz rate). For example, you do not need to be in high-level to use `InterconnectConfig.SetUARTConfiguration(uartConfig)`.

madan96, 2023-11-22:
Hi @odj3460 , you could try out our ROS HW interface [https://github.com/empriselab/kortex_hardware](https://github.com/empriselab/kortex_hardware) which provides support for effort mode at 1khz.

martinleroux, 2024-09-05:
Hi all,

I will be closing this issue by adding that our [ROS2 driver ](https://github.com/Kinovarobotics/ros2_kortex) does use low-level position control with MoveIT. For anyone needing real-time control through ROS, I recommend having a look at it.

I will also reiterate that real-time low-level control via ROS1 is not possible.

closed 2024-09-05 by martinleroux