Kinovarobotics/Kinova-kortex2_Gen3_G3L #179: Modbus Twist move type

TomSAVU, 2023-07-23:
Hi,
I am trying to use Modbus for obtaining a move similar to the angular moves in the Web app.
I tried to set the active move type to Twist, but I don't understand the effect of this setting.
Which data do I have to pass to which register for moving to a certain cartesian position or for setting the angular positions of the actuators?
Thanks!

felixmaisonneuve, 2023-07-24:
Hi @TomSAVU,

You cannot specify joint values and move the arm in angular position or velocity. 

What you can do instead is create some angular Actions with your angular positions/velocities and use modbus to call this Action usin its action_id

Best,
Felix

TomSAVU, 2023-07-24:
Thank you, Felix,
What exactly is the "Twist" active move type doing if I set both the linear positions and angular orientations different from the current ones and then start the Twist move?

felixmaisonneuve, 2023-07-24:
Twist commands are equivalent to "cartesian velocity". 
So you do not set linear position and angular orientation, you set linear velocities (m/s) and orientation velocities (deg/s).

For example, Twist command of [0;0;0.1;0;20;0] will move the arm in the Z axis at 0.1m/s and rotate the arm in the Y axis at 20deg/s

TomSAVU, 2023-07-24:
OK, clear now.
Will the angular movement be implemented for Modbus in the near future?
We have several cases in which we have to integrate the robot control into an application which has to use Modbus.

felixmaisonneuve, 2023-07-24:
As of now, it is not planned to implement angular control for modbus. 

I will leave this issue open and, if we plan on implementing this one day, I will update it.