Kinovarobotics/Kinova-kortex2_Gen3_G3L #180: Dual control using two KINOVA arms

tanrb, 2023-08-16:
## Summary

I am trying to connect to two kinova by connecting both ethernet cable from the kinova arms to an ethernet to usbc connector that is connected to a single port on my laptop. 

I have created two session managers, tcp and router client on python. I have also created the parseConnectionArguments2 to be used for the second args. However, I am unsure of how to change the ip address of one of the kinova to be different from the default '192.168.1.10' to connect to both robots at the same time.  Can I get some advice on how I should go about doing that?  Thanks!!!

kyilmazKinova, 2023-08-22:
Hi,
You can find the procedure to change the robot wired connection IP address in the Gen3's User Guide, page 74 to 75. The User Guide can be found here (https://www.kinovarobotics.com/product/gen3-robots#Product__resources) on the Kinova Website.

Best regards

felixmaisonneuve, 2023-09-20:
Hi @tanrb,

Did this solve your problem?
Can we close this issue?

Best,
Felix

tanrb, 2023-09-26:
Hi @felixmaisonneuve 

I am able to connect to both robots with different session managers, but I am just wondering how do I control both robots to move simultaneously rather than one after the other? 

Cheers

kyilmazKinova, 2023-11-08:
Hi, check out this issue https://github.com/Kinovarobotics/kortex/issues/45, @DjoleMNE provided a code example.

martinleroux, 2024-09-05:
I am closing this issue since the answer was given above. Before I do so however, I will add that Kortex API does not provide API endpoints that route commands to multiple robots at the same time. In short, the only way to synchronize robots is to leverage the fact that most commands are non-blocking (or rather, block until the robot acknowledges them, not until it is done moving), or to use low-level control with async calls in C++.

closed 2024-09-05 by martinleroux