Kinovarobotics/Kinova-kortex2_Gen3_G3L #181: Lost Connectivity

pouyan-asg, 2023-08-18:
## Description

I am using a Gen3 robot, and during work (without a specific reason), it lost connection. The error is "**An existing connection was forcibly closed by a remote host**," and it cannot make a connection again. In this case, I should connect through the webpage and start again. Also, always at the same waypoint, this problem happened. 
For example, in my case, exactly at the same Cartesian point.
I should tell you that waypoints are all inside the robot workspace, and I used the same dataset without problem. Moreover, IP configuration is fine based on Kinova documentation (IP address: 192.168.1.100 // Subnet: 255.255.255.0)

Do you know what the problem is? Is it related to waypoints themselves? Should I do anything to prevent in future?

Thank you very much for your help

felixmaisonneuve, 2023-09-20:
Hi @pouyan-asg 

Sorry for the delay.

We are aware of a bug that disconnects the API after multiple hours of use, but it can happen sooner in some case. However, this problem does not seem to be related to a specific waypoint. 

If you tested multiple times and the problem always occurs on the same waypoint, can you send us a minimal example that reproduces the issue so we could try on our side?

Best,
Felix

pouyan-asg, 2023-09-21:
Hi @felixmaisonneuve 

Thanks for reaching out to me with your valuable information.
Good to know about that.
Honestly, at that time, I could not repeat that error again. However, if next time it happens, I will send you waypoints here to check.

Thank you very much.

closed 2024-09-05 by martinleroux