Kinovarobotics/Kinova-kortex2_Gen3_G3L #189: (Torque) Calibration

sophielueth, 2023-11-08:
Dear Kinova Team,

we wanted to calibrate the torque sensors in our Kinova Gen3. Looking for the correct interface in the documentation, I found [this](https://docs.kinovarobotics.com/ref/autogen/Messages/ActuatorConfig.html#kortex_api.autogen.TorqueCalibration). I am wondering what the parameters `global_gain` (float), `global_offset` (float), `gain` (float [] of length 4) and `offset` (float [] of length 4) represent, that exist for each actuator (see e.g. [here](https://github.com/Kinovarobotics/kortex/issues/38). I assume the torque sensors to be strain gauges and therefore only expected one gain and one offset per sensor for a linear mapping.

Could you please elaborate on how these parameters are used and how they were identified? 

On another note: I also came across the class `CalibrationItem` [here](https://github.com/Kinovarobotics/kortex/issues/38). It contains `COGGING`, `MAGNETIC`, `MOTOR`and `POSITION_RANGE`. 
What parameters exactly are calibrated here?

Thank you!

kyilmazKinova, 2023-11-08:
Hi, you should use the WebApp. Simply position your Gen3 in a candle pose then, on the WebApp, navigate to "Configurations/Robot/Actuators/ActuatorX". For every Actuator, click on "Set Offset". 

Best regards,

sophielueth, 2023-11-08:
Hi kyilmazKinova,

Thank you for your fast answer!
I understand that clicking "set offset" will set the offset of the torque sensors (probably the TorqueOffset shown [here](https://github.com/Kinovarobotics/kortex/issues/38)). This limits calibration to one static parameter but does not allow me to influence the readings of the robot when under load (apart from that offset, which is added).

I want to understand to what extent I can adjust/calibrate the robot's sensors. The function `WriteTorqueCalibration` which is part of `ActuatorConfigClient` takes a `Torque Calibration` object as input. This object has the parameters `global_gain`, `global_offset`, `gain` and `offset`. 

Could you please elaborate on how these parameters are used and how they were identified?

(And I'm still interested in information about the class CalibrationItem [here](https://github.com/Kinovarobotics/kortex/issues/38).)

Best,

martinleroux, 2024-09-04:
Hi @sophielueth ,

Apologies for the very late response. The tracking of this issue was lost when Kevin left Kinova.

The Kortex API exposes some functions that are intended for internal use only (this is an issue we have solved for Link 6 but have yet to backport on Gen3). Torque calibration is one of them.

While it is technically possible to use the `WriteTorqueCalibration` function to overwrite our calibration, you would not be able to modify the model itself. I also assure you that the equipment we use in our production line is most likely more accurate with this model than what you would be able to obtain in a lab environment. For this reason, I strongly advise _against_ trying to modify the torque calibration.