Kinovarobotics/Kinova-kortex2_Gen3_G3L #191: update python version for the api wheel

patrickwasp, 2023-11-18:
## Summary

python 3.5.2 has reached end of life

## Use case

it's not straightforward to install old versions of python on ubuntu 22.04

martinleroux, 2024-03-19:
Hi all,

Our documentation is not up to date (we are working on that), but I can confirm Python 3.11 works fine.

I will leave this issue open for now in case you encounter some issue migrating.

Cheers

KarolineH, 2024-05-17:
Dear Kinova Team,
I am still getting an AttributeError when using Python 3.10 but not 3.8
Error: "module 'collections' has no attribute 'MutableMapping'" when trying to import anything from kortex_api
I'd like to avoid having to up/downgrade my Python version. Any other ideas?

martinleroux, 2024-05-17:
This issue actually lies in the version of the protobuf package that is installed with Python 3.10 or higher. 
You can try installing the .whl file with the following commands:

pip install protobuf==3.20 deprecated==1.2.7
pip install "kortex_api-2.6.0.post3-py3-none-any.whl" --no-deps

A PR should comme some to clarify this in the documentation.

closed 2024-05-17 by martinleroux

s-m-asjad, 2024-05-28:
I am not sure if this was resolved but adding the following lines before importing kortex_api solved it for me. 

```python
import sys
if sys.version_info.major == 3 and sys.version_info.minor >= 10:    # Change to == if it already works for 3.11
    import collections
    setattr(collections, "MutableMapping", collections.abc.MutableMapping)
    setattr(collections,"MutableSequence", collections.abc.MutableSequence)

#from kortex_api...
```

The attribute `MutableMapping` was moved to `collections.abc` in python 3.10

s-m-asjad, 2024-05-28:
Just double checked. It does not seem to work with 3.11.4 as well. 
Same issue as mentioned previously. 
It may not be ideal to downgrade protobuf. If the documentation needs more time to update, I can create a PR with the above lines added to the python API examples. 

P.S. this would be necessary for anyone using ROS2 Humble along with the Python API.