Kinovarobotics/Kinova-kortex2_Gen3_G3L #192: Inconsistency in End Effector Transformation Matrix Calculation

wxailwx, 2023-11-22:
## Description

We have discovered that the two methods for calculating the end effector transformation matrix, as provided in the user manual, yield inconsistent results in three-dimensional coordinates. Additionally, both methods produce values that differ from those displayed on the web application.

Question Regarding tool_pose_theta:
Furthermore, we would like to inquire about the specific representation of the tool_pose_theta. For instance, is it in the form of Roll-Pitch-Yaw (RPY) with the xyz order?
Thank you for your help!
## Version

At a minimum, provide the Kortex API and Kortex-enabled device versions.

KortexAPI :2.9.0

Kortex Device :Gen3 lite

## Code example
[image]
[image]

hermattt, 2023-12-30:
Hi, 

Table 92 has an error in "Frame 3 to frame 4" with a star, -0.0064 should be positive like in example without star. 
Did you also add gripper transformation matrix to get the same values as on web page? (The easiest way would be to "add" 0.12m to last matrix) 
[image]
So from -0.0615 to -0.1815
Check https://github.com/Kinovarobotics/kortex/issues/101 regarding tool_pose_theta question.

martinleroux, 2024-09-04:
Hi @wxailwx ,

I addition to the (correct) answer provided by @hermattt above, you should know that theoretical calculations based on the transformation matrices provided in the User Guide will never properly match what is reported by the Web App. The reason for this is that every arm is different due to assembly tolerances, and thus is calibrated before leaving our production facilities. The transformation matrices provided in the Web App correspond to the ideal robot with no tolerances.

If you wish to obtain the calibration data for your arm, please contact us at support@kinova.ca with the serial number of your arm (found on the sticker at the base of the robot - not in the Web App) and we can get it from our archives.