Kinovarobotics/Kinova-kortex2_Gen3_G3L #193: Problem in linking step when using EtherCatMaster and KortexAPI

SebastienCol, 2024-01-18:
## Description

Problem in linking step when using EtherCatMaster and KortexAPI. Suggests recompiling with -fPIE but this does not work. Error output below. This started happening when using SOEM for other uses. When linked with Kortex library, this problem started occurring.

## Version

Originally KortexAPI version 2.3.0 and now still not working with KortexAPI version 2.6.0.

## Code example

```
  /usr/bin/ld: /opt/Kortex/lib/release/libKortexApiCpp.a(KeEthercatMaster.cpp.o): relocation R_X86_64_32S against symbol `_ZTV23KeEthercatMasterFactory' can not be used when making a PIE object; recompile with -fPIE
  /usr/bin/ld: failed to set dynamic section sizes: bad value
  collect2: error: ld returned 1 exit status
 ```

## Expected behavior

dbsxogh09, 2025-03-18:
I'm struggling with similar issue. Any workarounds?

martinleroux, 2025-03-18:
Hi,

Gen3 does not support EtherCAT. KeEtherCATMaster seems to be an artifact from the original development period that was never cleaned up. I'll flag it to engineering and the dependency may be removed in a future update.

jplayan, 2025-03-19:
Good morning,

I'm working with Sebastien in the same working group. In our case we are not linking directly SOEM and Kortex but we have two different libraries, one with SOEM as a dependency, and another one with Kortex. Our issue occurs when linking these two targets to a new library that depends on both. In the end we found two "fixes" for this issue that might potentially be useful for your case too @dbsxogh09.

- The first one is changing the linking order between the targets. We were linking first the library with SOEM and then the one with Kortex which was triggering this problem. The issue got solved once we linked first with Kortex and then SOEM.

- Another solution is to handle the kortex dependency into the intermediate target as private (e.g. target_link_libraries(my_library PRIVATE ${Kortex_LIBRARIES}))

Of course, the best solution would be to remove the EtherCAT dependency from Kortex if not used as martinleroux mentions, but meanwhile I hope these workarounds help .

martinleroux, 2025-03-19:
Thanks for sharing. I have confirmed with Engineering that the EtherCAT dependency comes from a library that is only used inside the robot (and even then, it doesn't actually use EtherCAT), so a task was added to their backlog to generate a version of the API with only the external dependencies. I have however no timeline to give as of now. Considering we have just made an update to the API with the bump to 2.7, it might be a while.

If you are having the same issue as above, please comment in this thread with a +1.  The more people are asking for a fix, the faster it will get done.