Kinovarobotics/Kinova-kortex2_Gen3_G3L #194: Torque mode example issues

zongyaojin, 2024-03-18:
Hi guys! I'm using this API to test Kinova-gen3-7dof robot's torque control, I'm extending `api_cpp/examples/108-Gen3_torque_control/01-torque_control_cyclic.cpp` to make all joints running in torque mode. With the same setup (code below), joints 1, 3, 4, 5, 7 seem to to in torque mode with reasonable compliance. Yet, joints 2 and 6 seem to run in position and are very stiff.

Can you take a look and let me know what did I do wrong? Thanks!

```
bool example_cyclic_torque_control(k_api::Base::BaseClient* baseClient,
    k_api::BaseCyclic::BaseCyclicClient* baseCyclicClient,
    k_api::ActuatorConfig::ActuatorConfigClient* actuatorConfigClient)
{

  // Clearing faults
  try {
    baseClient->ClearFaults();
  } catch (...) {
    spdlog::error("Unable to clear robot faults");
    return false;
  }

  // Get actuator count
  unsigned int numActuators = baseClient->GetActuatorCount().count();

  std::vector<double> q(numActuators, 0);
  std::vector<double> qdot(numActuators, 0);

  std::vector<double> qErr(numActuators, 0);
  std::vector<double> qErrIntegral(numActuators, 0);

  //   0, 15, 180, 230, 360, 55, 90
  std::vector<double> qDes {30, 35, 150, 260, 10, 75, 100};
  std::vector<double> tau(numActuators, 0);

  k_api::BaseCyclic::Feedback baseFeedback;
  k_api::BaseCyclic::Command baseCommand;

  int timer_count = 0;
  int64_t now = 0;
  int64_t last = 0;

  std::cout << "Initializing the arm for torque control example" << std::endl;

  // Set the baseClient in low-level servoing mode
  auto servoModeInfo = k_api::Base::ServoingModeInformation();
  servoModeInfo.set_servoing_mode(k_api::Base::ServoingMode::LOW_LEVEL_SERVOING);
  baseClient->SetServoingMode(servoModeInfo);

  baseFeedback = baseCyclicClient->RefreshFeedback();

  // Initialize actuators data structure, which returns the initialized actuator pointer
  // The returned pointers could be used by the way to set the actuator to their current position to avoid a
  // Kinova internal following error
  for (unsigned int i = 0; i < numActuators; i++) {
    baseCommand.add_actuators()->set_position(baseFeedback.actuators(i).position());
  }

  // Set first actuator in torque mode now that the command is equal to measure
  auto ctrlModeInfo = k_api::ActuatorConfig::ControlModeInformation();
  ctrlModeInfo.set_control_mode(k_api::ActuatorConfig::ControlMode::TORQUE);
  for (int i = 1; i < 8; ++i) {
    actuatorConfigClient->SetControlMode(ctrlModeInfo, i);
    // auto mode = actuatorConfigClient->GetControlMode(i);
    // spdlog::warn("[{}] control mode is [{}]", i, mode.control_mode());
  }

  spdlog::info("Running torque control example for [{}] seconds.", kTimeDuration);

  // Real-time loop
  while (timer_count < (kTimeDuration * 1000)) {
    now = GetTickUs();

    if (now - last > 1000) {
      // Position command to first actuator is set to measured one to avoid
      // following error to trigger

      baseFeedback = baseCyclicClient->RefreshFeedback();

      double kp = 10;
      double kd = 10;

      // Incrementing identifier ensures actuators can reject out of time frames
      baseCommand.set_frame_id(baseCommand.frame_id() + 1);
      if (baseCommand.frame_id() > kFrameIdMax) baseCommand.set_frame_id(0);

      for (int i = 0; i < q.size(); ++i) {
        q[i] = baseFeedback.actuators(i).position();
        qdot[i] = baseFeedback.actuators(i).velocity();

        tau[i] = -kp * DegToRad(q[i] - qDes[i]) - kd * DegToRad(qdot[i] - 0);

        /**
         * [Notes from Kinova manual]
         * When command types other than Joint positions are sent to actuators, always include the position returned by
         * the latest actuator feedback in the command; actuator positions are used to monitor following errors.
         */

        baseCommand.mutable_actuators(i)->set_position(baseFeedback.actuators(i).position());
        baseCommand.mutable_actuators(i)->set_torque_joint(tau[i]);
        baseCommand.mutable_actuators(i)->set_command_id(baseCommand.frame_id());
      }

      if (timer_count % 1000 == 0) {
        spdlog::info("q  : {0:.2f}", fmt::join(q, ", "));
        spdlog::info("qd : {0:.2f}", fmt::join(qdot, ", "));
        spdlog::info("tau: {0:.2f}", fmt::join(tau, ", "));
        spdlog::warn("");
      }

      baseFeedback = baseCyclicClient->Refresh(baseCommand);

      timer_count++;
      last = GetTickUs();
    }
  }

  // Set first actuator back in position
  ctrlModeInfo.set_control_mode(k_api::ActuatorConfig::ControlMode::POSITION);
  for (int i = 1; i < 8; ++i) { actuatorConfigClient->SetControlMode(ctrlModeInfo, i); }

  // Set the servoing mode back to Single Level
  servoModeInfo.set_servoing_mode(k_api::Base::ServoingMode::SINGLE_LEVEL_SERVOING);
  baseClient->SetServoingMode(servoModeInfo);

  spdlog::info(
      "Torque control example done, resetting control mode to [POSITION], servo mode to "
      "[SINGLE_LEVEL_SERVOING], sleeping a little before exiting.");

  // Wait for a bit
  std::this_thread::sleep_for(std::chrono::milliseconds(1000));

  return true;
}
```

martinleroux, 2024-03-18:
Hi @zongyaojin ,

At first glance I cannot see anything deeply wrong with your code.
You can validate that the joints have turned to  torque control mode by using the GetControlMode function of the ActuatorConfigClient (I see that you already have, but your post does not indicate the output).

One thing I could see that could explain a stiff feeling would be the arbitrary tuning of 10/10 for your PD controller gains, especially for joints that are pulled by gravity which should probably use either a feedforward or an integral term.

zongyaojin, 2024-03-19:
Hi @martinleroux,

Thanks for your reply! I started with that example again and somehow it worked. I'm not sure, maybe my refactored code missed some intermediate bookkeeping steps. But it's strange, since the bookkeeping steps are all done in the same for loops.

However, I think the arbitrarily tuned control gains shouldn't be a problem. Since these control gains are so small, they should effectively only provide some damping. Like setting Kp to 0, and Kd to some small number.

Anyway, thanks for your help, I really appreciate it!