Kinovarobotics/Kinova-kortex2_Gen3_G3L #199: Get joint position or torques freezes for 10 sec randomly

MiguelGarciaUVa, 2024-07-04:
Hello,
I have a code where I am getting the position of the joints in a loop.
I am receiving data from the robot through the Refresh Feedback function, but suddenly it stops receiving that data for exactly 10 seconds and then continues as usual. This happens randomly (I think). We would like to know the cause of this or if you are aware that this is a bug and how to fix it. The same applies to the torques.
The servoing mode is set to single level servoing, instead of low level, because I want to move the robot in high level mode at the same time. I don't know if the problem can come from that.
If someone can help me, I would appreciate it.
Thanks in advance.

martinleroux, 2024-07-04:
Hello @MiguelGarciaUVa ,

There is no issue using RefreshFeedback in High level, as long as the BaseCyclic client is connected via a UDP transport to port 10001.

We are not aware of any known issue which would cause this. Can you launch your code with the Monitoring page of the WebApp opened in the Detailed tab and verify if the data is still updating there? The WebApp also uses RefreshFeedback() to fetch this data, so this would confirm whether your issue lies in the robot or in your code.

If the WebApp also freezes, I suggest reinstalling the firmware via the Upgrade page (find the latest firmware on our Website).
If not, we'll need to get more detail about your implementation to help you diagnose the issue.

Cheers

MiguelGarciaUVa, 2024-07-05:
Hi @martinleroux, 
Thanks for your reply. I have tried it and the WebApp does not freeze, so it is probably a bug in my code.
I think it freezes 10 secs because I am using:
```
base_feedback = self.SendCallWithRetry(self.base_cyclic.RefreshFeedback, 3)

 @staticmethod
    def SendCallWithRetry(call, retry,  *args):
        i = 0
        arg_out = []
        while i < retry:
            try:
                arg_out = call(*args)
                break
            except:
                i = i + 1
                continue
        if i == retry:
            print("Failed to communicate")
        return arg_out
```
So maybe it takes 10 seconds to try to communicate, then try again.
When it retries, it can usually do it.
But now, I don't know why the freeze is systematic. Every 90secs it can't communicate so it freezes 10 secs to do so. My loop is as follows:

```
def RunCyclic(self, t_sample, print_stats):
        self.cyclic_running = True
        print("Run Cyclic")
        sys.stdout.flush()
        cyclic_count = 0  # Counts refresh
        stats_count = 0  # Counts stats prints
        failed_cyclic_count = 0  # Count communication timeouts

        t_now = time.time()
        t_cyclic = t_now  # cyclic time
        t_stats = t_now  # print  time
        t_init = t_now  # init   time
        print("Running torque control example for {} seconds".format(self.cyclic_t_end))

        while not self.kill_the_thread:
            t_now = time.time()
            self.base_feedback = self.SendCallWithRetry(self.base_cyclic.RefreshFeedback, 3)
            # Cyclic Refresh
            if (t_now - t_cyclic) >= t_sample:
                t_cyclic = t_now
                # Position command to first actuator is set to measured one to avoid following error to trigger
                # Bonus: When doing this instead of disabling the following error, if communication is lost and first
                #        actuator continue to move under torque command, resulting position error with command will
                #        trigger a following error and switch back the actuator in position command to hold its position
                actuador1 = self.base_feedback.actuators[0].position
                actuador2 = self.base_feedback.actuators[1].position
                actuador3 = self.base_feedback.actuators[2].position
                actuador4 = self.base_feedback.actuators[3].position
                actuador5 = self.base_feedback.actuators[4].position
                actuador6 = self.base_feedback.actuators[5].position
                actuador7 = self.base_feedback.actuators[6].position
                
                cyclic_count = cyclic_count + 1
```
 To init the cyclic:

```
def InitCyclic(self, sampling_time_cyclic, t_end, print_stats):

        if self.cyclic_running:
            return True

        print("Init Cyclic")
        sys.stdout.flush()

        base_feedback = self.SendCallWithRetry(self.base_cyclic.RefreshFeedback, 3)
        if base_feedback: 
            self.base_feedback = base_feedback

            # Init command frame
            for x in range(self.actuator_count):
                self.base_command.actuators[x].flags = 1  # servoing
                self.base_command.actuators[x].position = self.base_feedback.actuators[x].position
                print("Actuador ", x, ": ", self.base_command.actuators[x].position)
                
            # Init cyclic thread
            self.cyclic_t_end = t_end
            self.cyclic_thread = threading.Thread(target=self.RunCyclic, args=(sampling_time_cyclic, print_stats))
            self.cyclic_thread.daemon = True
            self.cyclic_thread.start()
            return True

        else:
            print("InitCyclic: failed to communicate")
            return False
```

Sampling time is 0.002 and execution time is infinite (t_end = 0)

martinleroux, 2024-07-05:
Are you obtaining the same behaviour by running the defaut torque control example?

https://github.com/Kinovarobotics/kortex/blob/master/api_python/examples/108-Gen3_torque_control/01-torque_control_cyclic.py

MiguelGarciaUVa, 2024-07-05:
No, because they don't use RefreshFeedback() in the loop.
I tried to use RefreshFeedback() in that example and i obtain the same behaviour, yes.

martinleroux, 2024-07-05:
We will try to replicate the issue and get back to you.

martinleroux, 2024-07-08:
I modified the Torque example above to replace all torque commands with just RefreshFeedback. I ran the example for 30 minutes and could not replicate your issue. I did notice however that the delay would vary a lot depending on what else I would be doing with my computer at the same time. I am currently running the example using Python on a Windows 10 PC with a bunch of browser tabs open, but the worse delay I have seen is ~35ms.

Here is the full code if you want to try it:

#! /usr/bin/env python3

###
# KINOVA (R) KORTEX (TM)
#
# Copyright (c) 2019 Kinova inc. All rights reserved.
#
# This software may be modified and distributed
# under the terms of the BSD 3-Clause license.
#
# Refer to the LICENSE file for details.
#
###

###
# * DESCRIPTION OF CURRENT EXAMPLE:
# ===============================
# This example works as a simili-haptic demo.
#     
# The last actuator, the small one holding the interconnect, acts as a torque sensing device commanding the first actuator.
# The first actuator, the big one on the base, is controlled in torque and its position is sent as a command to the last one.
# 
# The script can be launched through command line with python3: python torqueControl_example.py
# The PC should be connected through ethernet with the arm. Default IP address 192.168.1.10 is used as arm address.
# 
# 1- Connection with the base:
#     1- A TCP session is started on port 10000 for most API calls. Refresh is at 25ms on this port.
#     2- A UDP session is started on port 10001 for BaseCyclic calls. Refresh is at 1ms on this port only.
# 2- Initialization
#     1- First frame is built based on arm feedback to ensure continuity
#     2- First actuator torque command is set as well
#     3- Base is set in low-level servoing
#     4- First frame is sent
#     3- First actuator is switched to torque mode
# 3- Cyclic thread is running at 1ms
#     1- Torque command to first actuator is set to a multiple of last actuator torque measure minus its initial value to
#        avoid an initial offset error
#     2- Position command to last actuator equals first actuator position minus initial delta
#     
# 4- On keyboard interrupt, example stops
#     1- Cyclic thread is stopped
#     2- First actuator is set back to position control
#     3- Base is set in single level servoing (default)
###

import sys
import os

from kortex_api.autogen.client_stubs.ActuatorConfigClientRpc import ActuatorConfigClient
from kortex_api.autogen.client_stubs.ActuatorCyclicClientRpc import ActuatorCyclicClient
from kortex_api.autogen.client_stubs.BaseClientRpc import BaseClient
from kortex_api.autogen.client_stubs.BaseCyclicClientRpc import BaseCyclicClient
from kortex_api.autogen.client_stubs.DeviceConfigClientRpc import DeviceConfigClient
from kortex_api.autogen.client_stubs.DeviceManagerClientRpc import DeviceManagerClient
from kortex_api.autogen.messages import Session_pb2, ActuatorConfig_pb2, Base_pb2, BaseCyclic_pb2, Common_pb2
from kortex_api.RouterClient import RouterClientSendOptions

import time
import sys
import threading

class TorqueExample:
    def __init__(self, router, router_real_time):

        # Maximum allowed waiting time during actions (in seconds)
        self.ACTION_TIMEOUT_DURATION = 20
[... 274 lines cut ...]
    args = utilities.parseConnectionArguments(parser)

    # Create connection to the device and get the router
    with utilities.DeviceConnection.createTcpConnection(args) as router:

        with utilities.DeviceConnection.createUdpConnection(args) as router_real_time:

            example = TorqueExample(router, router_real_time)

            success = example.InitCyclic(args.cyclic_time, args.duration, args.print_stats)

            if success:

                while example.cyclic_running:
                    try:
                        time.sleep(0.5)
                    except KeyboardInterrupt:
                        break
            
                example.StopCyclic()

            return 0 if success else 1

if __name__ == "__main__":
    exit(main())

MiguelGarciaUVa, 2024-07-09:
Thank you very much for your response and your time.

I ran it and the issue persists. This is the output console:

[image]

I am running it on Spyder on a Windows 11 PC. Did you change any parameters in utilities? Maybe there is a parameter causing the program to stop 10 seconds.
Today, for some reason, instead of freezing every 90 seconds, it runs for 30 seconds, stops, runs again for 60 seconds, stops, runs for 30 seconds, stops, and so on.
I noticed that it happens when it can't call the function RefreshFeedback() in SendCallWithRetry(). It's like it tries for 10 seconds to call it, until it realises it can't, and then it tries again, and succeeds. However, it has to wait 10 seconds before deciding that it couldn't and stops trying. It also happens without using SendCallWithRetry(), so maybe there is a parameter that sets this 10-second connection attempt. My API version is 2.6.0.
My program is this one, in case you want to try (I know there might be a lot of things I don't need, but it doesn't matter):

```
#! /usr/bin/env python3

##################################################################
# Programa encargado de leer las posiciones de las articulaciones
# y mandárselas a Unity
##################################################################
import sys
import os
import collections
try:
    from collections import abc
    collections.MutableMapping = abc.MutableMapping
    collections.MutableSequence = abc.MutableSequence
except:
    pass

from kortex_api.autogen.client_stubs.ActuatorConfigClientRpc import ActuatorConfigClient
from kortex_api.autogen.client_stubs.ActuatorCyclicClientRpc import ActuatorCyclicClient
from kortex_api.autogen.client_stubs.BaseClientRpc import BaseClient
from kortex_api.autogen.client_stubs.BaseCyclicClientRpc import BaseCyclicClient
from kortex_api.autogen.client_stubs.DeviceConfigClientRpc import DeviceConfigClient
from kortex_api.autogen.client_stubs.DeviceManagerClientRpc import DeviceManagerClient
from kortex_api.autogen.messages import Session_pb2, ActuatorConfig_pb2, Base_pb2, BaseCyclic_pb2, Common_pb2
from kortex_api.RouterClient import RouterClientSendOptions

import time
import sys
import threading
import socket
import struct

class ReadJoints:
    def __init__(self, router, router_real_time):

        # Create required services
        device_manager = DeviceManagerClient(router)
        
        self.actuator_config = ActuatorConfigClient(router)
        self.base = BaseClient(router)
        self.base_cyclic = BaseCyclicClient(router_real_time)

        self.base_command = BaseCyclic_pb2.Command()
        self.base_feedback = BaseCyclic_pb2.Feedback()
        self.base_custom_data = BaseCyclic_pb2.CustomData()

        # Detect all devices
        device_handles = device_manager.ReadAllDevices()
        self.actuator_count = self.base.GetActuatorCount().count

        # Only actuators are relevant
        for handle in device_handles.device_handle:
            if handle.device_type == Common_pb2.BIG_ACTUATOR or handle.device_type == Common_pb2.SMALL_ACTUATOR:
                self.base_command.actuators.add()
                self.base_feedback.actuators.add()

        # Change send option to reduce max timeout at 3ms
        self.sendOption = RouterClientSendOptions()
        self.sendOption.andForget = False
[... 238 lines cut ...]
        if (self.credentials[0] != ""):
            session_info = Session_pb2.CreateSessionInfo()
            session_info.username = self.credentials[0]
            session_info.password = self.credentials[1]
            session_info.session_inactivity_timeout = 10000   # (milliseconds)
            session_info.connection_inactivity_timeout = 2000 # (milliseconds)

            self.sessionManager = SessionManager(self.router)
            print("Logging as", self.credentials[0], "on device", self.ipAddress)
            self.sessionManager.CreateSession(session_info)

        return self.router

    # Called when exiting 'with' statement
    def __exit__(self, exc_type, exc_value, traceback):
    
        if self.sessionManager != None:

            router_options = RouterClientSendOptions()
            router_options.timeout_ms = 1000 
            
            self.sessionManager.CloseSession(router_options)

        self.transport.disconnect()
```

martinleroux, 2024-07-09:
I have not changed anything in the Utilities file. 
Have you tried reinstalling the firmware on your robot and performing a factory reset? 
Have you tried running the same code from another computer? I am tempted to blame either Windows 11 or your firewall configuration. I have seen somewhat similar issues in the past where the cause was that the network card of the PC could not handle all of the communication it had to (code + browser + other stuff running on the PC at the same time), which would cause some calls to just drop.

You can try to change your session_inactivity_timeout in Utilities to less than 10 seconds. I expect it will only reduce your systematic time to recall to whatever value you select without solving the root of the issue, but it may help you make progress.

martinleroux, 2024-09-04:
Hello @MiguelGarciaUVa ,
Any update on this topic? Can I close this issue?