Kinovarobotics/Kinova-kortex2_Gen3_G3L #201: calculation of Homogeneous Transformation matrice on 6DOF Gen3 robot

myhappygit, 2024-07-29:
## Description

Hello, 
I have been checking the robot pose goes well, and I found something strange with calculation of homogeneous transformation.
With the transformation matrice on the 6DOF Gen3 Document page 195-196, I have calculated position of the end-effector on the ZERO pose is [0,0, 0.001, 1.1765], but the KINOVA web app shows different values [0.005, 0.01, 1.176].
In addition, when I make a rotation with joint 2, 3, 5 respectively to move along x-axis of robot base frame, the translational motion of y-axis appears.
Please let me know what is correct and corrected values of transformation matrice. 

## Version

KortexAPI :  2.3.0-52

Kortex Device : 6DOF Gen3 

## Steps to reproduce

1. *step 1...* go to ZERO pose of KINOVA Web App "Actions" Tab
2. *step 2...* check cartesian position of xyz and rotate the joint 2,3,5 respectively and check the cartesian position value again.

## Code example

```
    # paste code here #
```

https://github.com/user-attachments/assets/0542a373-7649-4ceb-a65c-8faa88a39cf5

## Expected behavior

same values of position on ZERO pose and maintaining the y-axis value while the joint 2, 3, 5 are rotating from ZERO. 

## Any other information

video shows the y-axis value variation when the joint 2 rotates. 
The file size is too large to upload everything, but the same issue occurs with joints 3 and 5.

martinleroux, 2024-07-29:
Hello @myhappygit 

This is normal. All Gen3 units are calibrated to compensate for assembly tolerances. You can consider the Web App results to be more accurate than what you obtain from the "ideal" model given by the documentation.

If you need the calibration parameters specific to your unit, contact us at support@kinova.ca with the serial number of your arm (found on a sticker at the base of the robot - the one in the Web App is the SN of the controller only). We'll send you a copy of the calibration file of your robot from our archives.

myhappygit, 2024-07-30:
Thank you for your reply.
I'll try to send my robot information to support@kinova.ca

closed 2024-07-30 by myhappygit