Kinovarobotics/Kinova-kortex2_Gen3_G3L #202: Sequence Aborted with error 0:SUB_ERROR_NONE

Louis2099, 2024-08-25:
HI I'm student using kinova kortex python API on a gen3 7dof robot arm, my settings are listed as following:
kortex 2.6.0.post3
frimware 2.5.2-1

The problem I've been experiencing is that, I'm using the sequence example "api_python/examples/102-Movement_high_level/02-sequence.py" by adding a series of "move to Cartesian position" actions, following is one of my example trajectories:
"""
example_traj = ([0.28747064, 0.04720456, 1.03254818]), 
        np.array([0.40148122, 0.01917565, 0.96486813]), 
        np.array([ 0.52025528, -0.02537908,  0.86594063]), 
        np.array([ 0.63002006, -0.08680653,  0.72415356]), 
        np.array([ 0.70775769, -0.16291518,  0.53187928]), 
        np.array([ 0.72229969, -0.24377655,  0.29120655]), 
        np.array([ 0.55527246, -0.35700423,  0.15574953])
"""
But I always got error message that shows:
"Sequence aborted with error 0:SUB_ERROR_NONE"

I cannot find useful information to resolve the issue, my guess is that the trajectory has certain kind of limitations, thus not all action sequences can be executed?

I'd be appreciated if you can help resolve this issue.

martinleroux, 2024-08-26:
Hi @Louis2099 ,

There is currently a known bug which causes this that we unfortunately have yet to find a solution for, but we hope to solve for our next firmware release. In most cases, this happens at apparently random moments, after a few hours running. Is this the case for you? Does the error always seem to trigger at the same place in your sequence?

martinleroux, 2024-08-26:
What if you delete the very first point in your trajectory?
Also, are you sure the poses you are commanding are reachable by the robot? Is the gripper configured as installed? (this adds a Z component to the TCP which can make a pose reachable with a gripper but unreachable without).

What is the robot's starting pose when launching this sequence?
Can you play the trajectory backwards?

Louis2099, 2024-08-26:
The robot cannot execute individual waypoint, it shows executing action but never complete it, I assume the reason is the tool_pose and tool_theta cannot be met at the same time, I defined actions in following manner:
cartesian_pose = action.reach_pose.target_pose
    cartesian_pose.x = control[0]          # (meters)
    cartesian_pose.y = control[1]    # (meters)
    cartesian_pose.z = control[2]    # (meters)
    cartesian_pose.theta_x = control_theta[0] # (degrees)
    cartesian_pose.theta_y = control_theta[1] # (degrees)
    cartesian_pose.theta_z = control_theta[2] # (degrees)
in the example trajectory for the first waypoint theta is set to [179.44644165039062, -0.06476558744907379, 90.08074951171875], and position is [0.28747064, 0.04720456, 1.03254818]. 
if this is causing the stall, I wanna ask it there way that define a trajectory action sequence without explicitly define the tool_theta, or is there a tool_theta value that fits all position.