Kinovarobotics/Kinova-kortex2_Gen3_G3L #204: Actuator 5 Fault Error - Possible Bug

saulcarvalho, 2024-09-17:
## Description

I am using a Kinova Gen3 robot  which I have developed a large code which is operational, based on waypoint trajectories. I have an algorithm to generate the waypoint coordinates depending on the resolution that I want. Right now I am only using cartesian coordinates, and the movement is only in a plane in the Y and Z axes with the X axis value being constant. Thing is, for a trajectory whose waypoints are separated from 5mm or higher, it works well, but when I try to reduce it to 1mm that is where I start getting problems. 

The robot seems to be able to position itself fairly well at different coordinates positioned 1mm from each other, but after a few waypoint iterations (around 20 or 30 something), the robot enters a fault state saying the actuator 5 is in fault and that the motor current threshold was passed. When I do the same thing for 5mm, the full code can execute without a problem. Funny thing is, I am not even using actuator 5 during these specific movements ...

## Version

At a minimum, provide the Kortex API and Kortex-enabled device versions.

KortexAPI :

Kortex Device :

## Steps to reproduce

Executing the cartesian waypoint example from the kortex python API with a step in Y of 1mm between coordinates for 30-ish ierations will reproduce the same error.

## Expected behavior

Ideally I would want to do the trajectories without raising the error for a resolution of 1mm between coordinates.

martinleroux, 2024-09-17:
Hi @saulcarvalho ,

I will try to replicate your issue as soon as I can find some time. In the meantime, I want to highlight that waypoints are not really designed for densely-defined trajectories. Not only is the maximum number of waypoints in a list fairly low (1000), but dense trajectories also tend to have sharper changes in direction, which can slow the robot down a lot to ensure that the acceleration limits are always respected. For dense trajectories, we instead recommend either low-level position control, or servoing using Twist Commands.

That being said, none of these issues would explain why a joint fault would be triggered by such a trajectory, so I will get back to you once I have looked deeper into it.

Cheers

martinleroux, 2024-09-17:
I have just tried the following trajectory using the send_cartesian_waypoint_trajectory.py example (tested with and without optimal blending):

``` 
waypointsDefinition = ( (0.6,   0.0,  0.4,  0.0, kTheta_x, kTheta_y, kTheta_z),
                                    (0.6,  0.001, 0.4, 0.0, kTheta_x, kTheta_y, kTheta_z),
                                    (0.6,  0.002, 0.4, 0.0, kTheta_x, kTheta_y, kTheta_z),
                                    (0.6,  0.003, 0.4, 0.0, kTheta_x, kTheta_y, kTheta_z),
                                    (0.6,  0.003, 0.4, 0.0, kTheta_x, kTheta_y, kTheta_z),
                                    (0.6,  0.004, 0.4, 0.0, kTheta_x, kTheta_y, kTheta_z),
                                    (0.6,  0.005, 0.4, 0.0, kTheta_x, kTheta_y, kTheta_z),
                                    (0.6,  0.006, 0.4, 0.0, kTheta_x, kTheta_y, kTheta_z),
                                    (0.6,  0.007, 0.4, 0.0, kTheta_x, kTheta_y, kTheta_z),
                                    (0.6,  0.008, 0.4, 0.0, kTheta_x, kTheta_y, kTheta_z),
                                    (0.6,  0.009, 0.4, 0.0, kTheta_x, kTheta_y, kTheta_z),
                                    (0.6,  0.010, 0.4, 0.0, kTheta_x, kTheta_y, kTheta_z),
                                    (0.6,  0.011, 0.4, 0.0, kTheta_x, kTheta_y, kTheta_z),
                                    (0.6,  0.012, 0.4, 0.0, kTheta_x, kTheta_y, kTheta_z),
                                    (0.6,  0.013, 0.4, 0.0, kTheta_x, kTheta_y, kTheta_z),
                                    (0.6,  0.014, 0.4, 0.0, kTheta_x, kTheta_y, kTheta_z),
                                    (0.6,  0.015, 0.4, 0.0, kTheta_x, kTheta_y, kTheta_z),
                                    (0.6,  0.016, 0.4, 0.0, kTheta_x, kTheta_y, kTheta_z),
                                    (0.6,  0.017, 0.4, 0.0, kTheta_x, kTheta_y, kTheta_z),
                                    (0.6,  0.018, 0.4, 0.0, kTheta_x, kTheta_y, kTheta_z),
                                    (0.6,  0.019, 0.4, 0.0, kTheta_x, kTheta_y, kTheta_z),
                                    (0.6,  0.020, 0.4, 0.0, kTheta_x, kTheta_y, kTheta_z),
                                    (0.6,  0.021, 0.4, 0.0, kTheta_x, kTheta_y, kTheta_z),
                                    (0.6,  0.022, 0.4, 0.0, kTheta_x, kTheta_y, kTheta_z),
                                    (0.6,  0.023, 0.4, 0.0, kTheta_x, kTheta_y, kTheta_z),
                                    (0.6,  0.024, 0.4, 0.0, kTheta_x, kTheta_y, kTheta_z),
                                    (0.6,  0.025, 0.4, 0.0, kTheta_x, kTheta_y, kTheta_z),
                                    (0.6,  0.026, 0.4, 0.0, kTheta_x, kTheta_y, kTheta_z),
                                    (0.6,  0.027, 0.4, 0.0, kTheta_x, kTheta_y, kTheta_z),
                                    (0.6,  0.028, 0.4, 0.0, kTheta_x, kTheta_y, kTheta_z),
                                    (0.6,  0.029, 0.4, 0.0, kTheta_x, kTheta_y, kTheta_z),
                                    (0.6,  0.030, 0.4, 0.0, kTheta_x, kTheta_y, kTheta_z),
                                    (0.6,  0.031, 0.4, 0.0, kTheta_x, kTheta_y, kTheta_z)
                                    )
  ```

I was not able to replicate your issue even after replaying the trajectory a handful of times, but I think I have a pretty good idea as to what you are experiencing. As I mentioned above, Waypoints are really not designed for dense trajectories. At this level of density, the behaviour is even worse than I expected - to avoid overshooting points, the robot brakes at every waypoint and quickly starts moving again. In my setup, this induces a large amount of vibration in the robot. Depending on the orientation and the payload, this could easily trigger any joint fault between maximum velocity, maximum torque, maximum current or following error - current just happens to be the one that triggers first.

martinleroux, 2024-09-17:
Within 1mm of travel, the robot still won't near 10cm/s of speed, hence why you aren't seeing any improvement.

If you are stopping the robot completely at every point, you may consider using Sequences or Reach Pose actions (see examples 102/01 and 102/02). These use a somewhat less aggressive acceleration value and may help with the issue.  Try with velocity constraints around 0.01-0.05 m/s.

saulcarvalho, 2024-09-20:
Reach Pose actions is exactly what I am using. 

It seems that controling both joint speeds (around 20 to 30 degree/s) and linear speed (around 5 mm/s) to lower values improves the operation of the robot for some time, but after a few waypoints the same error happens. I don't think sequence actions are going to work well for the measurements that I'm doing, unfortunately.

martinleroux, 2024-09-20:
Oh, I assumed you used waypoints because you said so in your problem description and then closed the ticket since you only replied with a thumbs up to my last message.

If you are already using ReachPose, then Sequences won't improve the behaviour anyway. You could try using actual waypoints. The maximum acceleration is a bit higher but the acceleration profile is smoother (trapezoidal instead of bang-bang). If this still does not resolve your issue, the other alternatives I can think of are much more complicated, such as implementing a joint impedance controller to let J5 wiggle around its position when vibrated instead of locking it in place. Beyond that, I am afraid there isn't much I can do.

I will reopen the issue and leave it open until I hear back from you.

saulcarvalho, 2024-10-07:
I have been doing other things meanwhile, and I got to it again ... 

Resuming our earlier conversation:

I have changed this to an actual waypoint trajectory (I suppose you were talking about this). I am now able to run a trajectory until a certain time, then it throws the exact same error as with the ReachPose.

First thing I have noticed is the WaypointList only accepts up to 1000 coordinates, so what I did is I took chunks of the entire coordinate list (which I was using in the ReachPose, and I fed the ExecuteWaypointTrajectory function first with around 800 points at a time. Then I noticed the error occured after a chunk. I tried to reduce the number of coordinates in a chunk for 100 per chunk, and then it threw the same error after 4 or 5 chunks.

The thing is, even if this worked, using such a trajectory I cannot really make the measurements I want after each waypoint, unless I fork and change the original functions to stopafter a waypoint, which I was hoping I didn't have to do.

martinleroux, 2024-10-07:
Indeed, Waypoints lists only support up to 1000 waypoints - they are not well designed for dense trajectories, only better than ReachPose. Since your trajectory is dense, you could use a 3rd party kinematics library (I can recommend Peter Corke's Robotics Toolbox for Python) and command the robot via our low-level API.

Are you still observing the same error as you did with Reach Poses or is the robot reporting something else?

What exactly prevents you from taking measurements without stopping? Is it still vibration?

saulcarvalho, 2024-10-07:
I still get the exact same error as before, but the funny thing is, now it's on actuator 7 (before was always in actuator 5), I am using exactly the same area for the trajectory and the same movement resolution as before:
 - Error safety raised: Motor Current Mismatch
 - Error safety raised: Joint Fault

So with a trajectory, the robot reaches the points and it vibrates. I need to perform a measurement at exactly each point, thus I need to stop the robot before doing the measurement and then resume the trajectory.

I will try to follow the other suggestion you gave me.

martinleroux, 2024-10-07:
Can you tell me what your trajectory looks like? If we ignore the fact that you want to take measurements at specific positions, how what is the motion performed by the robot? Could the same trajectory be defined in a less dense manner by removing points that are co-linear? If so, it would be much easier to define the trajectory with the minimum number of waypoints and implement a mechanism to gather the data in parallel.

saulcarvalho, 2024-10-07:
Im going to respond to your questions one by one:

1. **Can you tell me what your trajectory looks like?** 
For now the movement is simply a plane movement in XZ, with Y fixed, and measurements are performed at every point in the plane, thus we obtain a matrix of waypoints of a plane. The movement is performed in a zig zag motion from -0.1 to 0.1 m in X and -0.1 to 0.1 in Z in relation to the home position (x = 0.794, y = 0, z = 0.536) with a movement resolution of 1 mm. The area of the measurement plane has been validated for other resolutions and it worked fine for ReachPose, the errors only started appearing below a 5 mm resolution for the measurement plane

2. **How what is the motion performed by the robot?** 
The motion performed by the robot is simply a reach pose to our predefined home for safety reasons, and then it jumps to the first measurement waypoint which is defined by (0.894, 0.0, 0.436, 0.0, 90.0, 0.0, 90.0), and starts decrementing X 1 mm by 1mm until reaching x = 0.694 after which it will move to z = 0.437, and move x back to 0.894, repeating this until x = 0.694 and z = 0.636

3. **Could the same trajectory be defined in a less dense manner by removing points that are co-linear?**
Perhaps it could, if only the first and last point of each line of the matrix are preserved, the complexity of the trajectory will greatly diminish. I would only need to guarantee that the movement from x_start to x_stop of each line of the matrix is performed 1 mm at a time and that the measurement is performed at the respective points. One of the solutions I was thinking was separate the movement of the robot and measurement algorithm in different threads and have them sinchronyze at each waypoint/pose. My question in this matter of minimizing waypoints is, if we perform a less dense movement trajectory, how do I guarantee the measurements at each coordinate? I know I can increase the duration of the movement and thus greatly slow the movement from x_start to x_stop, but this does not guarantee that I will be able to perform the measurements at each point.

martinleroux, 2024-10-07:
By zig zag you mean a large zig zag pattern, where all Y coordinates are covered before incrementing X (or vice-versa) and not something akin to a stitching patern? 
In other words, your trajectory generator looks like this? (I'll use Matlab notation for the sake of density)

```
for x = x0 : 0.001 : xf
    for y = y0 : 0.0001 : yf
        coordinate[end+1] = [x, y]
```
(Or you may have a mechanism in place to do back-and-forth instead of zig-zags)

In that case, I would say a sparse trajectory in Y is definitely the way to go - you wouldn't even really need to multi-thread to gather your data - although it would be a bit cleaner if you did. The trick is to use RefreshFeedback from a baseCyclic client running through a UDP transport. When used via UDP, RefreshFeedback can be called at a rate of 1 kHz and checked against your desired position for taking measurements (or you could gather ALL the measurements and then filter only the data that you want after the fact).

Here is some pseudo code illustrating what I mean:

```
trajectory = GenerateDenseTrajectory() # your existing trajectory
sparseTraj = GenerateSpareTrajectory() # Same trajectorybut with all co-linear consecutive points removed.

waypointList = ToWaypoints(sparseTraj) #formatting
baseClient.ExecutewaypointTrajectory(waypointList) # This function does not block while the robot is moving so you can perform your monitoring right away.

while TrajectoryIsNotFinished : #You can use ActionNotifications or just check if you reached the last point of the sparseTraj

    feedback = baseCyclicClient.RefreshFeedback()
    x = feedback.base.tool_pose_x
    y = feedback.base.tool_pose_y
    
    d = CalcDistanceToNextDensePoint(trajectory, [x,y])
    
    if d <tolerance:
        GetData() # Don't call RefreshFeedback again here or you'll lose 1ms.
        SetNextDensePoint() 
        
```

saulcarvalho, 2024-10-09:
So, I did what you said, seems to partially work but it has a lot of vibration and I cant garantee the exactitude of the measurements as the robot needs to be still when I perform them. I can see that the robot has the resolution to perform the trajectory movements as I debuged the calculated distance and it can do the readings below 1/10 mm.

Thing is, once again,  we have the same errors in joint 5 after some time of operation, so I don't think this has anything to do with the types of functions being used:
- Error safety raised: Motor Current Mismatch
- Error safety raised: Joint Fault

Best solution up until now in terms of stability was the first one we ever used, the Reach Pose one, because the arm reaches the pose, reads, and then moves to the next one and reads ...

Maybe we should make a like a dense trajectory and then feed the function with a single point and somehow terminate/reset the robot and then move to the next point. At the moment we don't really care about the time it takes to perform the whole measurement, we just want it to perform the measurements at the coordinates we want with the robot stopped.

martinleroux, 2024-10-09:
Admittedly at the moment I don't have a clue as to what causes the motor current mismatch. I was not able to replicate it here and the fact that it only seems to trigger while you're doing this specifically suggests that the hardware is not damaged. Just in case, can you run the Demo sequence (should be part of the default programs after a factory reset, but if you don't have it, you can build any dummy sparse traejectory with ~5-6 points) and let it run overnight? Just to confirm 100% that the hardware is fine.

I am surprised that the solution I suggested generates a lot of vibration. If you still want to investigate this, can you send a video to support@kinova.ca (mention you are sending it to Martin and the ticket will be assigned to me)? Otherwise, if you are satisfied with waiting at each point, I will close this issue.

saulcarvalho, 2024-10-10:
In response to the last comment, after nearly 10h of measurements, 10 x 10 cm with 1 mm resolution.

After all the tests that have been perforned so far, it seems for now it is a matter of plane area together with the resolution of the movements and also the fixed Y coordinate (as only X and Z are changing over time).

saulcarvalho, 2024-10-23:
We have been trying to solve the problem up until now, and we still have the same issue ... that is why I haven't closed the issue.

One thing that I also want to do, but documentation isn't exactly clear on that is:
- How do I know a specific safety was raised for an actuator? DeviceConfig's GetAllSafetyInformation doesn't seem to alter at all when these safeties are raised for either actuator 5 or 7 current mismatch. Basically I want to know if there's a safety raise for an actuator let's say 5 and if there is I want to specifically know which one(s).

```
# I've overly simplified the code a lot and summarised it as much as I can, specially for the measurement parts which do not 
# really matter for the movement in question
# The constants below come from an .ini file, I'm just adding their values below to understand where they come from in the function full_measurement_cycle()

TIMEOUT_DURATION = 999999
ANGULAR_JOINT_SPEEDS = [30, 30, 30, 30, 20, 20, 20]
N_RES = 0.001
REACHPOSE_WAITTIME = 1
MEASUREMENT_WAITTIME = 0.5

# Function that performs measurement
def full_measurement_cycle(base, base_cyclic, control_config, device_config):
    base_servo_mode = Base_pb2.ServoingModeInformation()
    base_servo_mode.servoing_mode = Base_pb2.SINGLE_LEVEL_SERVOING
    base.SetServoingMode(base_servo_mode)

    # Calculate trajectory - this gives the trajectory which goes line by line from bottom to top in a back and forth fashion; 
    # so basically all X coordinates are covered from left to right within a range and then Z is incremented and then right to left, 
    # repeating until it  finishes - once again, the trajectory works completely fine for N_RES > 0.005, without safety raises
    wpts = compute_2Dmatrix() 

    action = Base_pb2.Action()
    action.name = "<info> Movement is starting..."
    action.application_data = ""
    
    # Set the twist linear speed limit
    soft_limit = ControlConfig_pb2.TwistLinearSoftLimit()
    soft_limit.control_mode = ControlConfig_pb2.CARTESIAN_WAYPOINT_TRAJECTORY
    soft_limit.twist_linear_soft_limit = 0.1
    control_config.SetTwistLinearSoftLimit(soft_limit)
    # Set the twist angular speed limit
    soft_limit = ControlConfig_pb2.TwistAngularSoftLimit()
    soft_limit.control_mode = ControlConfig_pb2.CARTESIAN_WAYPOINT_TRAJECTORY
    soft_limit.twist_angular_soft_limit = 50.0
    control_config.SetTwistAngularSoftLimit(soft_limit)
    # Set the joint speed soft limits
    joint_limit = ControlConfig_pb2.JointSpeedSoftLimits()
    joint_limit.control_mode = ControlConfig_pb2.CARTESIAN_WAYPOINT_TRAJECTORY
    limit = 50.0
    joint_limit.joint_speed_soft_limits.extend([limit, limit, limit, limit, limit-10, limit-10, limit-10])
    control_config.SetJointSpeedSoftLimits(joint_limit)
    # Move to first position fast
    cartesian_pose = action.reach_pose.target_pose
    cartesian_pose.x = wpts[0][0] # (meters)
    cartesian_pose.y = wpts[0][1] # (meters)
    cartesian_pose.z = wpts[0][2] # (meters)
    cartesian_pose.theta_x = wpts[0][4] # (degree)
    cartesian_pose.theta_y = wpts[0][5] # (degree)
    cartesian_pose.theta_z = wpts[0][6] # (degree)
    
    e = threading.Event()
    notification_handle = base.OnNotificationActionTopic(check_for_end_or_abort(e), Base_pb2.NotificationOptions())
    base.ExecuteAction(action)
    finished = e.wait(TIMEOUT_DURATION)
    base.Unsubscribe(notification_handle)

    # Decrease the twist linear speed limit for the measurements
    soft_limit = ControlConfig_pb2.TwistLinearSoftLimit()
    soft_limit.control_mode = ControlConfig_pb2.CARTESIAN_WAYPOINT_TRAJECTORY
    soft_limit.twist_linear_soft_limit = N_RES*0.1
    control_config.SetTwistLinearSoftLimit(soft_limit)
    # Decrease the twist angular speed limit
    soft_limit = ControlConfig_pb2.TwistAngularSoftLimit()
    soft_limit.control_mode = ControlConfig_pb2.CARTESIAN_WAYPOINT_TRAJECTORY
    soft_limit.twist_angular_soft_limit = ANGULAR_JOINT_SPEEDS[0]
    control_config.SetTwistAngularSoftLimit(soft_limit)
    # Decrease the joint speed soft limits
    joint_limit = ControlConfig_pb2.JointSpeedSoftLimits()
    joint_limit.control_mode = ControlConfig_pb2.CARTESIAN_WAYPOINT_TRAJECTORY
    joint_limit.joint_speed_soft_limits.extend([ANGULAR_JOINT_SPEEDS[0], ANGULAR_JOINT_SPEEDS[1], ANGULAR_JOINT_SPEEDS[2], ANGULAR_JOINT_SPEEDS[3], ANGULAR_JOINT_SPEEDS[4], ANGULAR_JOINT_SPEEDS[5], ANGULAR_JOINT_SPEEDS[6]])
    control_config.SetJointSpeedSoftLimits(joint_limit)

    data = []
    for i in range(len(wpts)):
        action.name = f"Robot moving to position {i}"
        cartesian_pose = action.reach_pose.target_pose
        cartesian_pose.x = wpts[i][0] # (meters)
        cartesian_pose.y = wpts[i][1] # (meters)
        cartesian_pose.z = wpts[i][2] # (meters)
        cartesian_pose.theta_x = wpts[i][4] # (degree)
        cartesian_pose.theta_y = wpts[i][5] # (degree)
        cartesian_pose.theta_z = wpts[i][6] # (degree)
        
        e = threading.Event()
        notification_handle = base.OnNotificationActionTopic(check_for_end_or_abort(e), Base_pb2.NotificationOptions())
        base.ExecuteAction(action)
        finished = e.wait(TIMEOUT_DURATION)
        base.Unsubscribe(notification_handle)
        
        if DEBUG_FLAG:
              print(f"<DEBUG> Position({i}) = {wpts[i]}")   

        plot_traj(wpts[i], i,len(wpts))  # plot trajectory
        # pause to guarantee robot stabilization before measurement
        time.sleep(REACHPOSE_WAITTIME)      

        data[i] = get_data() # perform measurement at waypoint
        time.sleep(MEASUREMENT_WAITTIME)     

save_data_file(data) # save data to file
```

martinleroux, 2024-10-23:
If you detect a Joint Fault on the controller, you can have a look at the actuator feedback ( from baseCyclicClient.Feedback().actuators[i] ). The actuator feedback contains fields called fault_bank_a and fault_bank_b which contain Safety identifiers if faults are triggered. These IDs can be mapped to actual faults via the [SafetyIdentifierBankA enum ](https://github.com/Kinovarobotics/ros_kortex/blob/noetic-devel/kortex_driver/protos/ActuatorConfig.proto#L312-L343)

saulcarvalho, 2024-10-30:
So what I found out is that the joint are getting current spikes and they send the robot down, what I eventually did is to detect this make the robot stop some seconds and restart from last waypoint.

Did a long measurement for 20 + hours at 1mm resolution and everything is fine.

Im going to mark this as closed.

closed 2024-10-30 by saulcarvalho