Kinovarobotics/Kinova-kortex2_Gen3_G3L #205: How do we use the control config and communicate it to the robot? Ie., how do we change the joint acceleration limits?

taskseg, 2024-09-25:
## Description

One paragraph to explain the issue

## Version

At a minimum, provide the Kortex API and Kortex-enabled device versions.

KortexAPI :

Kortex Device :

## Steps to reproduce

1. *step 1...*
2. *step 2...*

## Code example

```
    # paste code here #
```

## Expected behavior

Describe the expected behavior (in regard to the bug described above).

## Any other information

Any other information you believe the developers need to know.

martinleroux, 2024-09-25:
Hi @taskseg ,
First, you have to create a Control Config client object, the same way you usually do a Base client object.
Then, you can simply call SetJointAccelerationSoftLimits(limits).

See [here](https://github.com/Kinovarobotics/ros_kortex/blob/noetic-devel/kortex_driver/protos/ControlConfig.proto#L65).

taskseg, 2024-09-26:
Thanks Martin! Is that how you would do it in python as well?

closed 2024-12-09 by martinleroux