Kinovarobotics/Kinova-kortex2_Gen3_G3L #207: How to control actuators with low level control in python

jes-bro, 2024-10-16:
Hi Martin, is there an example script that demonstrates how to use ActuatorCyclic/how to control actuators at the lower level that aren't the gripper? We're trying to independently control actuators at the 1kHz rate. Thanks!

jes-bro, 2024-10-16:
New question: how do you do low level position control in python?

martinleroux, 2024-12-09:
Hi @jes-bro ,

It seems the low-level position control example is missing in Python.
The logic is the same as in C++, so you can have a look at [this example](https://github.com/Kinovarobotics/Kinova-kortex2_Gen3_G3L/blob/master/api_cpp/examples/200-Actuator_low_level_control/01-actuator_low_level_velocity_control.cpp).

Alternatively, you can refer to the syntax in the [torque control example](https://github.com/Kinovarobotics/Kinova-kortex2_Gen3_G3L/blob/master/api_python/examples/108-Gen3_torque_control/01-torque_control_cyclic.py) and simply not switch the control mode to torque at line 195.

In short, the easiest way to do low-level position control is to create a BaseCyclic client object with a UDP transport and then call baseCyclicClient.Refresh(command) at 1 kHz.

It is possible to control the individual actuators through the ActuatorCyclic client by putting the robot in LOW_LEVEL_BYPASS mode instead of LOW_LEVEL_SERVOING, but I have yet to hear of any reason to do it since the result is just doing the above with extra steps (i.e. calling Refresh once per actuator instead of a single time for all of them every frame). As a matter of fact, I expect low-level via ActuatorCyclic not to be possible in Python not because the feature is not supported, but because it is unlikely that your system can hold a stable 1kHz control loop with 6-7 different devices at the same time in an interpreted language.