Kinovarobotics/Kinova-kortex2_Gen3_G3L #210: Forward Kinematics Mismatch using the official URDF

kczttm, 2024-11-19:
## Description

Hello, I'm interested in using the URDF file given [here](https://github.com/Kinovarobotics/ros_kortex/tree/kinetic-devel/kortex_description/arms/gen3) to do some dynamics simulation. However, I found small discrepancies (~2-10mm) between the resulted end-effector position `p_cal` and the API's ground truth `p_ee`.

We doubled checked to make sure that we are choosing the right model (7dof gen3, no vision, no gripper). Since the discrepancies are always small, I wonder if it comes from the fact that our arm is the fixed-based version (no blue base). 

Another important information is that as we place the arm at zero configuration, as shown in [here](https://github.com/Kinovarobotics/Kinova-kortex2_Gen3_G3L/issues/155#issuecomment-1466584984) or the page 258 of the Kinova® Gen3 Ultra lightweight robot user guide r9.1, the vertical number adds up to `1.187375` m which is closer to the third entry of `p_cal` below. This made me wonder if we are actually using the wrong URDF file since the beginning. 

Below are some of the results:
### zero configuration
```
q: [  0.0006   0.0005 359.9991   0.0001   0.0053 359.9997 359.9975]
p_ee: [ 0.0021  0.0002  1.1892 -1.5179 -0.0792  0.4818]
p_cal: [ 0.     -0.0249  1.1874  0.0004  0.0002 -0.0026]
```

### home configuration
```
q: [359.9969  15.0011 179.9982 229.9985   0.0058  55.0003  90.0009]
p_ee: [ 0.4556  0.0029  0.4295 90.9459  0.6073 89.5313]
p_cal: [ 0.4567  0.0014  0.4337 90.0022  0.0036 90.0095]

Thank you!

martinleroux, 2024-12-09:
Hi @kczttm ,

This discrepancy is expected. The URDF file is that of an ideal robot - extracted from the CAD, and does not take into consideration the tolerances of the assembly. Each unit is calibrated to adjust for these tolerances, so you can always expect the Web App to be returning a more accurate position than the URDF itself.

As I have mentioned in a few threads recently, if you need the calibration results for your robot, please contact support@kinova.ca with your serial number (found on the sticker at the base of the robot, WOXXXXXXX-X - do not send the SN found in the Web App since it corresponds to the controller's SN and is not what we are looking for).

kczttm, 2024-12-10:
Hello @martinleroux, thanks for the follow up. I have received the calibration xml file, and I briefly understood the math behind incorporating the calibration homogeneous matrix during each step of the forward kinematic iteration. Instead of doing this manually, I would like to know if there's a script that could help me incorporate the xml file on top of the original official URDF.

martinleroux, 2024-12-10:
While it is theoretically possible to apply the transforms to the URDF, I don't have such a script readily available to share since within the robot we are applying the calibration parameters during runtime.