Kinovarobotics/Kinova-kortex2_Gen3_G3L #214: Cannot compile C++ API

GlenHenshaw, 2025-02-05:
## Description

Following the instructions on the website to the letter on Ubuntu results in the following error:

    > ./scripts/build-gcc debug

    conanfile.txt: Installing package
    Requirements
        kortex_api_cpp/2.6.0-r.3@kortex/stable from 'kinova_public' - Cache
    Packages
        kortex_api_cpp/2.6.0-r.3@kortex/stable:227cd850c3226dee2456c1f7b318da6887458d58 - Missing

    Installing (downloading, building) binaries...

    ERROR: Missing binary: kortex_api_cpp/2.6.0-r.3@kortex/stable:227cd850c3226dee2456c1f7b318da6887458d58
    kortex_api_cpp/2.6.0-r.3@kortex/stable: WARN: Can't find a 'kortex_api_cpp/2.6.0-r.3@kortex/stable' package for the specified settings, options and dependencies:
    - Settings: arch=armv8, build_type=Release, compiler=gcc, compiler.libcxx=libstdc++11, compiler.version=5, os=Linux
    - Options: shared=False
    - Dependencies: 
    - Requirements: 
    - Package ID: 227cd850c3226dee2456c1f7b318da6887458d58

    ERROR: Missing prebuilt package for 'kortex_api_cpp/2.6.0-r.3@kortex/stable'
    Use 'conan search kortex_api_cpp/2.6.0-r.3@kortex/stable --table=table.html -r=remote' and open the table.html file to see available packages
    Or try to build locally from sources with '--build=kortex_api_cpp'

    More Info at 'https://docs.conan.io/en/latest/faq/troubleshooting.html#error-missing-prebuilt-package'

    CMake Error at thirdParty/conan.cmake:399 (message):
      Conan install failed='1'
    Call Stack (most recent call first):
      thirdParty/conan.cmake:485 (conan_cmake_install)
      CMakeLists.txt:123 (conan_cmake_run)

## Version

At a minimum, provide the Kortex API and Kortex-enabled device versions.

KortexAPI : 2.6.0

Kortex Device : ???

## Steps to reproduce

Follow all the steps in the document

    https://www.thekitchn.com/best-costco-products-2-23520449

## Expected behavior

I expect the code to build without errors when following the instructions in the build document.

## Any other information

This build is running in an Ubuntu 22.04 VM on an M3 Macintosh inside QEMU

GlenHenshaw, 2025-02-06:
Hahaha. Yes, I did link the wrong document; now fixed.

martinleroux, 2025-02-06:
```kortex_api/kortex_api_cpp/2.6.0-r.3@kortex/stable``` is a path to our Artifactory server, not a local path to your computer.

I think the error lies in the path at line 120 of the CmakeLists file. I can find the Conan package on our Artifactory server [here](https://artifactory.kinovaapps.com/ui/repos/tree/General/conan-local-public/kortex/kortex_api_cpp/2.6.0-r.3/stable/76518ae0f34da66238e0ae668d275bf053cc5342), but the URL does not seem to match the CMake file. I don't really understand how such a problem could have occured, as far as I know we haven't changed the Artifactory structure recently.

First thing I would try is to simply change line 120 to ```URL https://artifactory.kinovaapps.com/artifactory/api/conan/conan-local-public)```. If this still does not work, revert to the original plan to bypass Conan, but don't forget to set the ```USE_CONAN``` variable to false. If you have already downloaded the API files and put them in the proper repos (following the tutorial), you can also put ```DOWNLOAD_API``` to false (see lines 43 and 44).

I don't have time to go through the process myself right now, but I'll add to my team's todo list to check that the CMakefile instructions for Conan are actually functional, and update the repo should we find it is not.