Kinovarobotics/Kinova-kortex2_Gen3_G3L #215: About low-level servoing (or bypass ervoing) and possibile soulutions

pos03052, 2025-02-06:
### Summary
To enable real-time control, we have been transitioning from high-level control to low-level control over the past few weeks. We successfully implemented our own forward and inverse kinematics calculations, applied the actual URDF of our robot, and confirmed that it moves correctly.

### Issue: Uncontrollable Joint Speed and Acceleration
However, we encountered a critical issue: in low-level control, As far as I know, velocity and acceleration of individual joints cannot be modified. As a result, when following a trajectory, the robot moves too fast. 

My prevoius issue about low-level control modulation:

- [https://github.com/Kinovarobotics/Kinova-kortex2_Gen3_G3L/issues/211](url)

So, at high speeds, motion appears natural.
At low speeds, the robot reaches each target point at 1 kHz, causing a stop-and-go effect.
This makes smooth motion difficult, especially for precise and slow movements.

### Potential Workarounds and Limitations
One possible workaround is to interpolate the trajectory or joint targets into smaller steps. This could partially mitigate the issue, it still does not provide smooth, general-purpose robot control. Instead, the behavior becomes more similar to a stepper motor, which is not ideal for many applications.

### Questions and Alternative Solutions
Given these limitations, I would like to ask:

Are there any alternative solutions to this issue?
Would low-level bypass control be a viable way to resolve this?
From my understanding, low-level bypass control needs direct control of all seven actuators at 1 kHz, but I am unsure whether joint velocity and acceleration can be adjusted in this mode. If there is a way to achieve smooth, precise motion control at low speeds, I would appreciate any guidance on this.

pos03052, 2025-02-13:
Thank you for your detailed explanation. I understand what you mean.

I believe the communication cycle was 1 kHz, but I will first check a few things on my end.

I plan to verify the target trajectory, the inverse kinematics results (i.e., each joint angles), and the communication cycle. I suspect there might be some discontinuities in these values.
(This is because my understanding is that if the trajectory, joint angles, and control frequency are continuous at 1 kHz, the robot should move smoothly—even if the target trajectory moves very slowly.)

I don't think I will try low-level bypass for now as you recommended.

So, I will check for any discontinuities, and if I find any, I will correct them. If not, I will send you my code and a video.

Additionally, in low-level mode, can I expect the motor drive to generate a smooth trajectory under control of the maximum velocity and acceleration of the 7 actuators?

Thank you.

martinleroux, 2025-02-13:
Yes. Be wary not too flirt too much with the limit though. If you run into Following Errors, it means your command is getting ahead of the actuator. The tolerance is pretty high (~5° if I remember correctly) but it can accumulate if you are just slightly over.